OpenDroneMap/ODM #1072: ODM fails to send remote tasks when using --sm-cluster and the --crop float argument

linusmartensson, 2020-02-25:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 79 on Linux

### What is the problem?

LRE fails to parse the input argument "--crop 3" correctly. With it set and using an --sm-cluster scheme, ODM fails to parse arguments in get_submodel_args_dict:
```
[INFO]    LRE: About to process submodel_0001 remotely
[WARNING] LRE: submodel_0001 failed with: 'float' object has no attribute 'startswith'
[INFO]    LRE: Re-queueing submodel_0001 (retries: 1)
```

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

The processing should proceed as normal

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Run any setup triggering the LRE in a remote split execution, for example with a single node connected on localhost:3000:
```
docker run --network="host" -ti --rm opendronemap/nodeodm 
docker run --network="host" -ti --rm -v ~/dataset:/datasets/code opendronemap/odm --project-path="/datasets" --crop 3 --split 50 --split-overlap 50 --sm-cluster http://localhost:3000
```

pierotofy, 2020-02-25:
Thanks @linusmartensson  :hand: good catch! This should have been fixed in #1073. Once docker images are rebuilt, update and see if it fixes the problem for you?

closed 2020-02-25 by pierotofy

linusmartensson, 2020-02-26:
Seems to be working now! :)

commit 75841ef1 (2020-02-25): Merge pull request #1073 from OpenDroneMap/croplrefix

Fix #1072