OpenDroneMap/ODM #1078: Only one partial reconstruction per submodel is aligned during align_submodels

linusmartensson, 2020-03-02:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 79 on Linux

### What is the problem?

When partial reconstructions are generated, there seems to be an issue in the reconstruction alignment output.
According to the OpenSfM sources, each partial reconstruction from each submodel is thrown into a large bundle adjustment problem, then output as a set of similarity transforms which are applied to each partial reconstruction and then saved to the aligned reconstruction file.
In reality however, it seems like only one partial reconstruction is updated. I can't seem to figure out why.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

I would expect each and every point and shot in every partial reconstruction to be modified at least at the 1e-6 magnitude, but this only seems to be true for a single partial reconstruction in each submodel. All other partial reconstructions are identical in a side-by-side comparison for each shot and point.

(As a sidenote, I'd be interested in understanding why the similarity transform is generated to create a rigid transform, rather than just applying the bundle output to the reconstruction as a whole. I would expect this to create smoother boundaries between submodels? Outlier handling?)

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Trigger the align_submodels stage in OpenSfM on a dataset split into multiple submodels which also contains multiple partial reconstructions. This should be easily achievable by lowering the minimum feature count and setting a split on a dataset with a sufficient number of entries.

linusmartensson, 2020-03-03:
I think this might be a bug rather than an enhancement. 
From the sources, all partial reconstructions should always be aligned, but I can't necessarily say it's even the index 0 partial we use in split/merge right now that's actually updated in the alignment - which implies that we might in these cases see a merge resulting in unaligned clouds.

linusmartensson, 2020-03-03:
Found it. Apparently the itertools.groupby method used in apply_transformations(transformations) in OpenSfM has to be passed an already sorted list, or it won't group entries correctly. Since the reconstruction is loaded for each group, the unaligned reconstructions will be reloaded and overwrite any previously aligned components.

Submitting a PR to 099 and mapillary OpenSfM in a bit. :tada:

linusmartensson, 2020-03-03:
https://github.com/OpenDroneMap/OpenSfM/pull/5 :dancer:

pierotofy, 2020-03-03:
Fantastic! Thanks for looking into this @linusmartensson

linusmartensson, 2020-03-03:
No prob! It's a fun codebase. :gift: 
Now merged in both OpenDroneMap/OpenSfM and mapillary/OpenSfM, so closing. :)

closed 2020-03-03 by linusmartensson

teiminryu, 2022-09-08:
Could you please tell me how to solve this issue correctly because my submodels still cannot be aligned?  Thank you so much.