OpenDroneMap/ODM #1085: Overflow with larger image sizes with odm_orthophoto

smathermather, 2020-03-17:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Natively

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

irrelevant

### What is the problem?

Line 512 of modules/odm_orthophoto/src/OdmOrthoPhoto.cpp causes issues with sufficiently large images, e.g. 52700 x 45976 but not e.g. 49510 x 38436.

With the following message:

```
Couldn't allocate enough memory to render the orthophoto (50191x43787 cells = 8790853268 bytes). Try to increase the --orthophoto-resolution parameter to a larger integer or add more RAM.
Traceback (most recent call last):
  File "run.py", line 57, in <module>
    app.execute()
  File "/home/odm/ODM/stages/odm_app.py", line 92, in execute
    self.first_stage.run()
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 445, in run
    self.process(self.args, outputs)
  File "/home/odm/ODM/stages/odm_orthophoto.py", line 89, in process
    '-outputCornerFile {corners} {bands}'.format(**kwargs))
  File "/home/odm/ODM/opendm/system.py", line 76, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Big images: good.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Any sufficiently high-resolution dataset of large enough extent should trigger. Can share dataset offline if necessary.

smathermather, 2020-03-17:
The threshold is suspiciously near 2000000000....

pierotofy, 2020-03-17:
It could be a problem of insufficient memory; how much RAM have you allocated (and is this a multispectral dataset?)

ivangayton, 2020-03-17:
Hi Piero, it's Ivan (it's my dataset that Steve is helping manage). It's a Digital Ocean droplet with 182 GB of RAM and another 192 GB of swap.

Standard Ebee photos from the SODA camera (RGB).

ivangayton, 2020-03-17:
@pierotofy can share access to the machine if poking around it might shed light on the problem (that would also give you access to the dataset and console logs; we ran ODM using nohup and saved console logs).

pierotofy, 2020-03-17:
Hey @ivangayton :hand: access to the machine would make it easier yes.

SSH key: https://gist.github.com/pierotofy/02f30a3ff2dcab7128042d4e7e32407c

ivangayton, 2020-03-17:
Done, added you. Access by ssh odm@dumditty.com

smathermather, 2020-03-17:
> Hi Piero, it's Ivan (it's my dataset that Steve is helping manage). It's a Digital Ocean droplet with 182 GB of RAM and another 192 GB of swap.
> 
> Standard Ebee photos from the SODA camera (RGB).

Actually, I changed that to 300GB swap... . :)

pierotofy, 2020-03-17:
What was the command (how was ODM invoked) that caused the error? (I see it's a split-merge workflow, so not sure which submodel failed).

pierotofy, 2020-03-17:
Probably not;

```
Material 79 rendered.
Material 80 rendered.
Material 81 rendered.
... model rendered

Writing ortho photo to /mnt/odmphotos/BackupMsimbazi/submodels/submodel_0000/odm_orthophoto/odm_orthophoto_render.tif
Writing corner coordinates to /mnt/odmphotos/BackupMsimbazi/submodels/submodel_0000/odm_orthophoto/odm_orthophoto_corners.txt
Orthophoto generation done.
```

Let me know how to reproduce it.

ivangayton, 2020-03-17:
No, that one succeeded! The original command was ```Currently running:

nohup python run.py BackupMsimbazi --split 0 --split-overlap 0 --ignore-gsd --depthmap-resolution 1000 --orthophoto-resolution 5 --dem-resolution 15 --pc-las --dsm &```.

That ran until submodel_0000, which crashed.
Error was ```
Feck.

[INFO]    Finished odm_dem stage
[INFO]    Running odm_orthophoto stage
[INFO]    running /home/odm/ODM/build/bin/odm_orthophoto -inputFiles /mnt/odmphotos/BackupMsimbazi/submodels/submodel_0000/odm_texturing_25d/odm_textured_model_geo.obj -logFile /mnt/odmphotos/BackupMsimbazi/submodels/submodel_0000/odm_orthophoto/odm_orthophoto_log.txt -outputFile /mnt/odmphotos/BackupMsimbazi/submodels/submodel_0000/odm_orthophoto/odm_orthophoto_render.tif -resolution 20.0  -outputCornerFile /mnt/odmphotos/BackupMsimbazi/submodels/submodel_0000/odm_orthophoto/odm_orthophoto_corners.txt 
Couldn't allocate enough memory to render the orthophoto (52700x45976 cells = 9691740800 bytes). Try to increase the --orthophoto-resolution parameter to a larger integer or add more RAM.
Traceback (most recent call last):
  File "/home/odm/ODM//run.py", line 57, in <module>
    app.execute()
  File "/home/odm/ODM/stages/odm_app.py", line 92, in execute
    self.first_stage.run()
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 445, in run
    self.process(self.args, outputs)
  File "/home/odm/ODM/stages/odm_orthophoto.py", line 89, in process
    '-outputCornerFile {corners} {bands}'.format(**kwargs))
  File "/home/odm/ODM/opendm/system.py", line 76, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
Traceback (most recent call last):
  File "run.py", line 57, in <module>
    app.execute()
  File "/home/odm/ODM/stages/odm_app.py", line 92, in execute
    self.first_stage.run()
  File "/home/odm/ODM/opendm/types.py", line 464, in run
    self.next_stage.run(outputs)
  File "/home/odm/ODM/opendm/types.py", line 445, in run
    self.process(self.args, outputs)
  File "/home/odm/ODM/stages/splitmerge.py", line 150, in process
    system.run(" ".join(map(quote, argv)), env_vars=os.environ.copy())
  File "/home/odm/ODM/opendm/system.py", line 76, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1```

ivangayton, 2020-03-17:
So to reproduce @pierotofy try running submodel_0000 at resolution 5

smathermather, 2020-03-17:
Correct. Run at 5.5 cm and it completes. Run at <5.4 cm and it fails quite quickly.

pierotofy, 2020-03-17:
The fix in #1086 is already applied to the machine. Double-check if it works, then close?

Thanks!

(I was able to process it):

```
Material 79 rendered.
Material 80 rendered.
Material 81 rendered.
... model rendered

...
Orthophoto generation done.
```

smathermather, 2020-03-18:
Reviewed the patch. This looks good.

PR #1086 (2020-03-18): Array overflow fix
Should fix #1085 

Thanks for the report @ivangayton @smathermather  :+1: 

Implicit cast was overflowing the integer multiplication (width * height) before casting to a larger data type.
closed 2020-03-18 by smathermather