OpenDroneMap/ODM #1095: Supplying a gcp_list.txt with WGS84 projection results in IndexError

Jessendelft, 2020-03-30:
I am supplying a gcp_list.txt file to my dataset, which looks as follows:

```
WGS84	
10.605415604158175	59.89444589880294	43.84716082433367	800.0	600.0	snapshot_2020_03_27_12_07_42_428_UTC_metadata.jpeg
10.605414955747381	59.89466090702028	45.87591677661027	800.0	600.0	snapshot_2020_03_27_12_07_59_228_UTC_metadata.jpeg
10.605414026300236	59.89487596982595	48.932939038801365	800.0	600.0	snapshot_2020_03_27_12_08_15_928_UTC_metadata.jpeg
10.605414494154864	59.89509104442486	49.9331889682139	800.0	600.0	snapshot_2020_03_27_12_08_32_726_UTC_metadata.jpeg
10.605843857218774	59.89508710166794	49.93194690049768	800.0	600.0	snapshot_2020_03_27_12_08_52_826_UTC_metadata.jpeg
10.605843971673016	59.89487128361989	49.93218518022779	800.0	600.0	snapshot_2020_03_27_12_09_09_725_UTC_metadata.jpeg
10.605844557311832	59.89465593333066	49.93251164224006	800.0	600.0	snapshot_2020_03_27_12_09_26_625_UTC_metadata.jpeg
10.605844120282839	59.8944404293301	49.932931421603044	800.0	600.0	snapshot_2020_03_27_12_09_43_525_UTC_metadata.jpeg
10.605843981388029	59.89422557134116	49.93343876752445	800.0	600.0	snapshot_2020_03_27_12_10_00_324_UTC_metadata.jpeg
10.605844618216405	59.89401019306841	49.93403589521486	800.0	600.0	snapshot_2020_03_27_12_10_17_224_UTC_metadata.jpeg
10.606273728381757	59.8940139427698	49.93288163208692	800.0	600.0	snapshot_2020_03_27_12_10_37_324_UTC_metadata.jpeg
10.60627293154728	59.8942297605667	49.93228457764073	800.0	600.0	snapshot_2020_03_27_12_10_54_123_UTC_metadata.jpeg
10.60627257808002	59.89444487253431	49.93177822445881	800.0	600.0	snapshot_2020_03_27_12_11_11_023_UTC_metadata.jpeg
10.606272460439827	59.894660319520085	49.931360658763964	800.0	600.0	snapshot_2020_03_27_12_11_27_822_UTC_metadata.jpeg
10.6062727865843	59.89487549035276	49.931032567608995	800.0	600.0	snapshot_2020_03_27_12_11_44_622_UTC_metadata.jpeg
10.606702027705195	59.89487149116057	49.92997189679109	800.0	600.0	snapshot_2020_03_27_12_12_04_821_UTC_metadata.jpeg
10.606700993088053	59.89465567546936	49.9303033734524	800.0	600.0	snapshot_2020_03_27_12_12_21_721_UTC_metadata.jpeg
10.606702684550642	59.89444068279106	49.93071695917284	800.0	600.0	snapshot_2020_03_27_12_12_38_720_UTC_metadata.jpeg
10.606701520786688	59.89422519349078	49.9312284725562	800.0	600.0	snapshot_2020_03_27_12_12_55_620_UTC_metadata.jpeg
10.606701152260156	59.894010317074915	49.931826557147446	800.0	600.0	snapshot_2020_03_27_12_13_12_519_UTC_metadata.jpeg
10.606702666333732	59.89379502846066	49.932511363644934	800.0	600.0	snapshot_2020_03_27_12_13_29_419_UTC_metadata.jpeg
10.607132141544835	59.89379845855679	49.931537531659544	800.0	600.0	snapshot_2020_03_27_12_13_49_518_UTC_metadata.jpeg
10.60713108651463	59.89401424538565	49.930850342004305	800.0	600.0	snapshot_2020_03_27_12_14_06_418_UTC_metadata.jpeg
10.607129695280147	59.89380053255691	42.068203345440494	800.0	600.0	snapshot_2020_03_27_12_19_00_645_UTC_metadata.jpeg
10.607131147556007	59.89401468887452	43.07081809289373	800.0	600.0	snapshot_2020_03_27_12_19_17_444_UTC_metadata.jpeg
10.607130245123711	59.89422958094833	44.07550905807217	800.0	600.0	snapshot_2020_03_27_12_19_34_343_UTC_metadata.jpeg
10.607130874155853	59.89444490215273	44.07498692427028	800.0	600.0	snapshot_2020_03_27_12_19_51_243_UTC_metadata.jpeg
10.607130847125156	59.89465961017429	46.1429740613022	800.0	600.0	snapshot_2020_03_27_12_21_20_641_UTC_metadata.jpeg
```

When running this set with the corresponding pictures, I get the following error:
```
[INFO]    Parsing SRS header: WGS84
Traceback (most recent call last):
File "/code/run.py", line 61, in <module>
app.execute()
File "/code/stages/odm_app.py", line 92, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 325, in run
self.process(self.args, outputs)
File "/code/stages/dataset.py", line 110, in process
rerun=self.rerun())
File "/code/opendm/types.py", line 72, in georeference_with_gcp
gcp = GCPFile(gcp_file)
File "/code/opendm/gcp.py", line 13, in __init__
self.read()
File "/code/opendm/gcp.py", line 23, in read
self.srs = location.parse_srs_header(self.raw_srs)
File "/code/opendm/location.py", line 123, in parse_srs_header
if ref[0] == 'WGS84' and ref[1] == 'UTM':
IndexError: list index out of range
```

This makes sense, as `ref = header.split(' ')` (File "/code/opendm/location.py", line 121) only gives back 1 argument, not 2. This looks like a bug to me.

pierotofy, 2020-03-30:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

(WGS84 is not a valid SRS header. Try `EPSG:<code>`).

closed 2020-03-30 by pierotofy

Jessendelft, 2020-03-30:
Dear Pierotofy,
If I considered this to be a question rather than a bug, I would definetly have put it there. If you say the header is wrong, the [documentation](https://github.com/mapillary/OpenSfM/blob/master/doc/source/gcp.rst) (which is referenced by the [OpenDroneMap Documentation](https://docs.opendronemap.org/tutorials.html?highlight=gcp#ground-control-points) multiple times) should be updated, as the example there states I can write WGS84 as a header.

smathermather, 2020-03-30:
Cool. @Jessendelft, can you open an issue here: https://github.com/OpenDroneMap/docs/issues

Bonus points for submitting a patch do docs!