OpenDroneMap/ODM #1102: Localization

KovalevArtem, 2020-04-21:
It is sad that the service is only in English - I propose to translate.
I am ready to help in the translation into Russian.
(For example, you can use Weblate, Crowdin, or any other service. Free for open source projects.)

pierotofy, 2020-04-21:
Agree. It's not just a matter of translating though, the code needs to be modified to support localization (via po files or similar). We'd welcome efforts toward that (open a pull request).

KovalevArtem, 2020-04-21:
There is a person from Weblate who can help with the integration of their service ...
@orangesunny

comradekingu, 2020-05-13:
+1 for Weblate. Crowdin isn't gratis for libre software, and is closed source.

KovalevArtem, 2020-07-28:
https://twitter.com/danbjoseph/status/1288064483106213891
It is a pity that the choice fell on Transifex. It's much more convenient to use Weblate...

pierotofy, 2020-07-28:
It wouldn't have been my choice either, but I appreciate that somebody has picked up the initiative to translate the docs!

comradekingu, 2020-07-28:
Transifex is primarily a exploitation and tracking platform.
As per usual with projects that have a CoC, the hypocrisy starts at line one.
>As contributors and maintainers, we pledge to respect all people who contribute …

pierotofy, 2020-07-28:
@comradekingu perhaps you could help setup a localization effort in weblate.

comradekingu, 2020-07-28:
Weblate has just the thing called "Hosted" for that sort of thing, and it is actually gratis for libre software.
https://hosted.weblate.org/hosting/ :)

pierotofy, 2020-07-28:
I've forwarded a request to translate WebODM.

[image]

danbjoseph, 2020-08-11:
there was a volunteer who was actively translating the ODM documentation ([repo](https://github.com/OpenDroneMap/docs) / [website](https://docs.opendronemap.org/)) by manually editing .po files like [this one](https://github.com/OpenDroneMap/docs/blob/publish/source/locale/sw/LC_MESSAGES/installation.po). managing that process, especially when adding more languages and editing the source text, was going to be a nightmare. sphinx-doc is being used to build the ODM docs and they [conveniently describe](https://www.sphinx-doc.org/en/master/usage/advanced/intl.html) how to push/pull translations to/from transifex. i didn't know about weblate.  this issue wasn't on the ODM docs repo so i didn't see it when i quickly created a process to manage the docs translation. now that i've heard about it, i'm interested in seeing how weblate might be incorporated. probably going to take some time though. links to applicable examples or detailed descriptions of how weblate would be integrated with the ODM docs would be welcome.

the ODM ecosystem includes a number of other pieces including the ODM software itself and WebODM. i'm sure there are translation needs for those as well and contributions would be welcome for setting those up.

KovalevArtem, 2020-08-12:
@orangesunny Need help!

orangesunny, 2020-08-12:
Hi all, we will be happy to host ODM on Hosted Weblate. Let’s solve some obstacles to do it!

1. I received a request for hosting WebODM from @pierotofy , mentioning .po, but only contains `nested JSON`s so far
[image]

2. @danbjoseph You can check how Weblate is familiar with Sphinx docs while reading Weblate of Godot Engine docs, both being translated by the community in Weblate.

It is not hard to set up, we only need to know that there are properly set up localization files in ODM repo and which one you want to translate... Or I can set up your project and leave the components on you. What do you prefer?

pierotofy, 2020-08-12:
For WebODM, we don't have the .po files yet (we're still working to export them). I can let you know once we do?

orangesunny, 2020-08-12:
> For WebODM, we don't have the .po files yet (we're still working to export them). I can let you know once we do?
> Sure you can. Once you have internationalization ready (files for translation), we can start localization (set up Weblate). Don’t have to be .po, there is a pallet of formats supported by Weblate: https://docs.weblate.org/en/latest/formats.html Make the choice that works the best for you!

> It would be awesome to have hosting through Weblate.
Our pleasure :)

danbjoseph, 2020-08-13:
@orangesunny thank you for the example, that's very helpful. i'll need some time to explore their [workflow](https://github.com/godotengine/godot-docs-l10n#how-to) and see how to adapt it to the ODM docs.

the ODM documentation and WebODM are parts of the same ecosystem but being managed in different repositories and different format of translation files. would it be structured on weblate as an "opendronemap" project, with then a "WebODM" component and a  "docs" component under that project?

smathermather, 2020-08-13:
> Transifex is primarily a exploitation and tracking platform.
> As per usual with projects that have a CoC, the hypocrisy starts at line one.
> 
> > As contributors and maintainers, we pledge to respect all people who contribute …

I want to address this very specifically, and as founder of the project, co-founder of the ecoystem (ht: @pierotofy) make clear that at the core of what we do in the OpenDroneMap ecosystem is the following:

> Awesome. Drone. Software.
> We’re creating the most sustainable drone mapping software with the friendliest community on earth.

We take the friendly side of this very seriously. The community is the most important part. The code and ecosystem are an artifact of that. The code can and will evolve, but the community needs to be a welcoming place for most. This means:

* There is a code of conduct. Violation of the code of conduct has consequences. The TLDR for any good CoC is that bullies aren't allowed. They restrict the speech of others. The rest is in the CoC. I suggest reading it and taking it to heart. If you can't take it to heart, this might not be the right place for you.
* [Optimistic merging](http://hintjens.com/blog:106) is practiced here with some of the following precepts:
  * People before code: build the right community and it will build the right code.
  * Progress before consensus: look for processes that work without upfront consensus (except on rules).
  * Problems before solutions: use a problem-driven process.
  * etc.

Escalation either in the issues or on Twitter isn't acceptable. We are all FOSS junkies here. We know how work gets done:
1. There needs to be an issue or specific set of issues relevant to the project at hand.
1. The issue or issues need(s) a champion either with the labor to complete the work, or the funding to requisition the work of others
1. Given the optimistic merging principles, we take the contributions of solutions that solve a real problem offered by those in the community who are willing to offer.

On twitter, I offered that the above is the needful. Thanks for those who hopped into this thread to give specifics on ways forward. Extra thanks to Dan for 
1. Settings up docs.opendronemap.org in the first place
1. Getting our translations on a new path with infrastructure
1. Being the first to [open a specific issue](https://github.com/OpenDroneMap/docs/issues/69) on how to move the needle forward on transitioning the infrastructure to something FOSS.

All the best,
Steve

orangesunny, 2020-08-13:
> @orangesunny thank you for the example, that's very helpful. i'll need some time to explore their [workflow](https://github.com/godotengine/godot-docs-l10n#how-to) and see how to adapt it to the ODM docs.

If you have any further questions about Sphinx docs and Weblate, hit us on the e-mail support@weblate.org ;) Our devs can share some wisdom 😸 

> 
> the ODM documentation and WebODM are parts of the same ecosystem but being managed in different repositories and different format of translation files. would it be structured on weblate as an "opendronemap" project, with then a "WebODM" component and a "docs" component under that project?

yes, you can do this structure. The only thing that is mandatory is that the code and translations of libre projects have to be in the same repository to be transparent and fair to the translators' community :) But components within one project can be from different repos.

pierotofy, 2020-12-14:
The first few strings are online! https://hosted.weblate.org/projects/webodm/ (Sorry it took a while). Effort is underway to get everything ready for translation at https://github.com/OpenDroneMap/WebODM/pull/943

orangesunny, 2020-12-20:
Welcome to the lovely Weblate community! Share the [webODM engage page](https://hosted.weblate.org/engage/webodm/) everywhere so you attract a crowd of new translators :)

pierotofy, 2020-12-20:
Thanks @orangesunny !

I'm really liking the platform; it has everything that we need and the barrier to contribute is really low (excellent!)

luzpaz, 2023-09-15:
Is there a reason ticket still remains open ?

comradekingu, 2023-09-15:
@luzpaz By the looks of it, no.
Happy to contribute if the CoC is dropped. :)

pierotofy, 2023-09-15:
We have localization in WebODM. Localization in ODM is possible also, but would require a major contribution.

I'm going to close this for now. if there's interest in localizing ODM also we'd welcome a pull request. :pray:

closed 2023-09-15 by pierotofy