OpenDroneMap/ODM #1111: Improved Resume Support

dawonn-haval, 2020-05-22:
### How did you install OpenDroneMap? (Docker, natively, ...)?

git clone <snip>
./webodm.sh start

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

chrome, Ubuntu

### What is the problem?

I ran out of disk space, and want to resume where it left off after making space. 

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

The ability to 'restart from' is great but I don't know what steps have been completed and where to restart from. I wish there was some way to find out, or for it to figure it out automatically. 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Process a large dataset on a small hard disk.

pierotofy, 2020-05-23:
Hey @dawonn-haval :hand: if your process was interrupted, you actually do not need to pass any parameters to ODM (perhaps your questions was about WebODM?). ODM will automatically figure out what parts are left to be processed if you start the pipeline again.

dawonn-haval, 2020-05-23:
yes, I was referring to webodm. :)

closed 2020-05-23 by dawonn-haval