OpenDroneMap/ODM #1113: Optimize

pierotofy, 2020-05-30:
`distance_transform_edt` calls are particularly memory inefficient and can lead failure. https://github.com/OpenDroneMap/ODM/blob/d4ea030d79cb3fdea2c3db64bf8418b3c8255aaf/opendm/orthophoto.py#L90

```
File "/code/stages/odm_orthophoto.py", line 149, in process
blend_distance=20, only_max_coords_feature=True)
File "/code/opendm/orthophoto.py", line 90, in compute_mask_raster
dist_t = ndimage.distance_transform_edt(alpha_band)
File "/usr/local/lib/python2.7/dist-packages/scipy/ndimage/morphology.py", line 2194, in distance_transform_edt
dt = ft - numpy.indices(input.shape, dtype=ft.dtype)
MemoryError
```

We should either process the array in chunks or write our own distance_transform_edt function.

Related: https://stackoverflow.com/questions/16925596/how-can-i-use-scipy-to-do-a-memory-efficient-distance-transform-operation

pierotofy, 2020-09-09:
Anothe report: https://github.com/OpenDroneMap/ODM/issues/1113

pierotofy, 2020-10-22:
This has been solved.

closed 2020-10-22 by pierotofy

ESimonson95, 2024-07-22:
hello, can someone direct me to the solution for distance_transform_edt memory errors? I am not using ODM but developing a workflow that requires using the function with larger raster arrays.

smathermather, 2024-08-09:
@ESimonson95 -- the following is the pull request that I think fixed it. I didn't do the work, so I cannot advise beyond that:
https://github.com/OpenDroneMap/ODM/pull/1156