OpenDroneMap/ODM #1115: Running ./start-dev-env.sh throws multiple Import errors

SBCV, 2020-06-03:
### How did you install OpenDroneMap? 
I followed the steps in https://github.com/OpenDroneMap/ODM#developers, i.e.
- `git clone https://github.com/OpenDroneMap/ODM.git`
- `cd` to cloned dir
- `DATA=/path/to/datasets ./start-dev-env.sh`

### What's your operating system? 
OS: Ubuntu 18.04

### What is the problem?

When executing the last step described under https://github.com/OpenDroneMap/ODM#developers (i.e. `./run.sh --project-path /datasets mydataset`) I receive the following error message:
`ImportError: No module named cv2`

I am able to fix this problem by manually installing OpenCV, i.e. 
- detaching from the container with `CTRL-p CTRL-q`
- logging in as root with `docker exec -it <container_id> bash`
- running `apt-get install python-opencv`
- detaching from the container with `CTRL-p CTRL-q`

When I attach to the container again with `docker attach <container_id>` and execute (again) `./run.sh --project-path /datasets mydataset` I receive:
`ImportError: No module named opensfm.large`. 
Thus, I think something went wrong with the installation process. 

Here is the log of `DATA=/path/to/datasets ./start-dev-env.sh`
```
Starting development environment...
Datasets path: /home/bullin/Desktop/empty
NodeODM port: 3000
QT Creator: NO
access control disabled, clients can connect from any host
Adding seb to /etc/passwd
Adding seb to /etc/group
su: Authentication failure
(Ignored)
bash: cannot set terminal process group (-1): Inappropriate ioctl for device
bash: no job control in this shell

###################################
ODM Dev Environment Ready. Hack on!
###################################
```

### What should be the expected behavior? 
Creating the same/similar results as executing 
`docker run -ti --rm -v /datasets:/datasets opendronemap/odm --project-path /datasets project `
as described under https://github.com/OpenDroneMap/ODM#quickstart

### How can we reproduce this?
Follow the steps at https://github.com/OpenDroneMap/ODM#developers

pierotofy, 2020-06-03:
Mm, really strange; are you using the `opendronemap/odm` docker image?

SBCV, 2020-06-03:
Side note: The second last line of `start-dev-env.sh` contains  `docker run -ti --entrypoint bash --name odmdev -v $(pwd):/code -v "$DATA":/datasets -p $PORT:3000 --privileged -e DISPLAY -e LANG=C.UTF-8 -e LC_ALL=C.UTF-8 -v="/tmp/.X11-unix:/tmp/.X11-unix:rw" -v="$HOME/.odm-dev-home:/home/$USER" opendronemap/nodeodm -c "/code/start-dev-env.sh --setup $USER $USER_ID $GROUP_ID $QTC", which contains `opendronemap/nodeodm`. 
That looks a bit strange.

pierotofy, 2020-06-03:
Using `opendronemap/nodeodm` is expected (it's built from `opendronemap/odm`). The opencv error usually shows up if the PYTHONPATH is not properly set to point to the OpenSfM directory (https://github.com/OpenDroneMap/ODM/blob/master/Dockerfile#L69).

SBCV, 2020-06-03:
Do I have to set that path by hand?

SBCV, 2020-06-03:
Btw: `/code/SuperBuild` in the docker container contains only `CMakeLists.txt` file and a `cmake` folder.

SBCV, 2020-06-03:
Did I miss some steps maybe?

pierotofy, 2020-06-03:
`/code/SuperBuild` should definitely have more files. Remove all of your docker images, then re-pull `opendronemap/nodeodm`.

SBCV, 2020-06-03:
I'll check, if it works and open a PR

SBCV, 2020-06-03:
Just to avoid any confusion:
I've now deleted all docker containers with `docker rm -vf $(docker ps -a -q)` and images with `docker rmi -f $(docker images -a -q)`

And next I should execute 
`(odmdev) [user:/code] master+* ± bash configure.sh reinstall`
before
`(odmdev) [user:/code] master+* ± ./run.sh --project-path /datasets mydataset`?

pierotofy, 2020-06-03:
Correct!

SBCV, 2020-06-03:
Nice, that worked. 
Just one more question: Is it possible to change the  data path set with 
`DATA=/path/to/datasets ./start-dev-env.sh` 
after executing `(odmdev) [user:/code] master+* ± bash configure.sh reinstall`?
If not we could add an additional note to the PR.

pierotofy, 2020-06-03:
Mm, probably not; it involves mounting a docker volume, which requires the container to be restarted.

PR #1116 (2020-06-03): Added missing configure step to readme.md
This resolves #1115
closed 2020-06-03 by pierotofy