OpenDroneMap/ODM #1130: Blank Orthophoto and map tiles for very close up images of ground field (~0.05 cm/pixel GSD)

theiosdevguy, 2020-07-01:
Dear Community,

I'd like to start with thanking everyone in making such a wonderful project. Thank you very much.

### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker (NodeODM)

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome, Ubuntu 18.04

### What is the problem?

Ground close up images with GSD around 0.05cm/pixel are not creating any orthophoto output and no tiles output. But 3D model is generated fine. 

I have a set of around 5000 images that are very close images of the ground. All of them have EXIF data with GPS coordinates in them. I tried with a sample of first 100, first 50 and even first 10 images but I do not get any output in orthophoto and tiles. The files are generated but are blank. I changed tiles creation zoom level to 30 (35 level throws exception) just to check if it needs more zoom level but it didn't change anything.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Maybe I am missing some configuration but does ODM create orthomosaics and tiles for very close up images of ground? Is there any way to achieve this?

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I am attaching first 10 images from my dataset
https://drive.google.com/file/d/1l3X7yVD6XuNg1g-laLBsKb9Uhsuz-mdW/view?usp=sharing

pierotofy, 2020-07-07:
I think you need to set `--ignore-gsd` and manually pass a better `--orthophoto-resolution` value. The defaults are not going to work well because ODM assumes that you are flying at a certain altitude.

theiosdevguy, 2020-07-09:
@pierotofy Thanks for the response. I tried with Orthophoto Resolution set to 100 and Ignore GSD set to TRUE but same results. Any idea why this would be happening?

pierotofy, 2020-07-09:
You probably need to lower the value, not increase it. Let's move the conversation to the forum though, as this looks like a usage question and not a bug :pray: https://community.opendronemap.org

closed 2020-07-09 by pierotofy

smathermather, 2020-07-09:
Psst, try to set orthophoto resolution to 0.05 and set ignore gsd, and yeah, come on over to community. It's a great place with lots of people who can be benefactors and beneficiaries of these support questions. In fact, if this works, do a quick post on community of what your question was and the solution.

theiosdevguy, 2020-07-09:
@pierotofy @smathermather Thanks. BTW, I tried 0.05 and ignore gsd but same results. No orthophoto, no tiles. But in all the cases I get 3D Model generated. I wonder if 3D Model is generated why is Orthophoto blank.

smathermather, 2020-07-09:
Linking to the community post from here as well: https://community.opendronemap.org/t/blank-orthophoto-and-map-tiles-for-very-close-up-images-of-ground-field-0-05-cm-pixel-gsd/4862/2