OpenDroneMap/ODM #1143: Processing mutlispectral dataset

octopuscabbage, 2020-08-11:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker, latest. 

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Command Line & Ubuntu 16.04.6 LTS

### What is the problem?

I created a 6 band tiff from a senterra 4 band camera. The first three bands are RGB, the other 3 bands are hyperspectral bands. 

The camera saves these as separate files in separate folder so I used python & tifffile to compile them into one 6 band tiff. (Side Note: is this correct?) This 6 band tiff opens fine in QGiS.   The file that it chokes on is attached but I don't think this is the only picture it would choke on. 

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

A multispectral tiff should be produced.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Attempt to run odm with this image as the input image. 

https://drive.google.com/file/d/1RFVchi8DyF_xBFua3ROyrvykpcexA2c0/view?usp=sharing

I hope google doesn't mangle my tiff.

octopuscabbage, 2020-08-11:
Is the problem that I need to add an alpha channel between the visible and non-visible channels?

octopuscabbage, 2020-08-11:
```

[INFO]    Initializing ODM - Mon Aug 10 23:57:38  2020

[INFO]    ==============

[INFO]    build_overviews: False

[INFO]    camera_lens: auto

[INFO]    cameras: {}

[INFO]    crop: 3

[INFO]    debug: False

[INFO]    dem_decimation: 1

[INFO]    dem_euclidean_map: False

[INFO]    dem_gapfill_steps: 3

[INFO]    dem_resolution: 5

[INFO]    depthmap_resolution: 640

[INFO]    dsm: False

[INFO]    dtm: False

[INFO]    end_with: odm_report

[INFO]    fast_orthophoto: False

[INFO]    feature_type: sift

[INFO]    force_gps: False

[INFO]    gcp: None

[INFO]    gps_accuracy: 15

[INFO]    ignore_gsd: False

[INFO]    matcher_distance: 0

[INFO]    matcher_neighbors: 8

[INFO]    max_concurrency: 24

[INFO]    merge: all

[INFO]    mesh_octree_depth: 10

[INFO]    mesh_point_weight: 4

[INFO]    mesh_samples: 1.0

[INFO]    mesh_size: 200000

[INFO]    min_num_features: 8000

[INFO]    mve_confidence: 0.6

[INFO]    name: code

[INFO]    opensfm_depthmap_method: PATCH_MATCH

[INFO]    opensfm_depthmap_min_consistent_views: 3

[INFO]    opensfm_depthmap_min_patch_sd: 1

[INFO]    optimize_disk_space: False

[INFO]    orthophoto_compression: DEFLATE

[INFO]    orthophoto_cutline: False

[INFO]    orthophoto_no_tiled: False

[INFO]    orthophoto_png: False

[INFO]    orthophoto_resolution: 5

[INFO]    pc_classify: False

[INFO]    pc_csv: True

[INFO]    pc_ept: False

[INFO]    pc_filter: 2.5

[INFO]    pc_las: True

[INFO]    pc_rectify: False

[INFO]    pc_sample: 0

[INFO]    project_path: /

[INFO]    radiometric_calibration: none

[INFO]    rerun: None

[INFO]    rerun_all: False

[INFO]    rerun_from: None

[INFO]    resize_to: 2048

[INFO]    skip_3dmodel: False

[INFO]    sm_cluster: None

[INFO]    smrf_scalar: 1.25

[INFO]    smrf_slope: 0.15

[INFO]    smrf_threshold: 0.5

[INFO]    smrf_window: 18.0

[INFO]    split: 999999

[INFO]    split_multitracks: False

[INFO]    split_overlap: 150

[INFO]    texturing_data_term: gmi

[INFO]    texturing_keep_unseen_faces: False

[INFO]    texturing_nadir_weight: 16

[INFO]    texturing_outlier_removal_type: gauss_clamping

[INFO]    texturing_skip_global_seam_leveling: False

[INFO]    texturing_skip_hole_filling: False

[INFO]    texturing_skip_local_seam_leveling: False

[INFO]    texturing_skip_visibility_test: False

[INFO]    texturing_tone_mapping: none

[INFO]    time: False

[INFO]    use_3dmesh: False

[INFO]    use_exif: False

[INFO]    use_fixed_camera_params: False

[INFO]    use_hybrid_bundle_adjustment: False

[INFO]    use_opensfm_dense: False

[INFO]    verbose: True

[INFO]    ==============

[INFO]    Running dataset stage

[INFO]    Loading dataset from: /code/images

[INFO]    Loading 51 images

[WARNING] Cannot read /code/images/a8166fce0145-FW00019.tiff with PIL, fallback to cv2: cannot identify image file '/code/images/a8166fce0145-FW00019.tiff'

imread_('/code/images/a8166fce0145-FW00019.tiff'): can't read header: OpenCV(3.4.6) /code/SuperBuild/src/opencv/modules/imgcodecs/src/grfmt_tiff.cpp:148: error: (-215:Assertion failed) channels <= 4 in function 'normalizeChannelsNumber'

Traceback (most recent call last):

  File "/code/run.py", line 62, in <module>

    app.execute()

  File "/code/stages/odm_app.py", line 95, in execute

    self.first_stage.run()

  File "/code/opendm/types.py", line 331, in run

    self.process(self.args, outputs)

  File "/code/stages/dataset.py", line 89, in process

    photos += [types.ODM_Photo(f)]

  File "/code/opendm/photo.py", line 68, in __init__

    self.parse_exif_values(path_file)

  File "/code/opendm/photo.py", line 231, in parse_exif_values

    self.width, self.height = get_image_size.get_image_size(_path_file)

  File "/code/opendm/get_image_size.py", line 19, in get_image_size

    width = img.shape[1]

AttributeError: 'NoneType' object has no attribute 'shape'
```

octopuscabbage, 2020-08-11:
Nevermind, I was going about this all wrong. For future reference: You put all of the images seperately in the same folder and odm will detect it's a multi camera setup.

closed 2020-08-11 by octopuscabbage

smathermather, 2020-08-24:
Thanks for posting the follow up!