OpenDroneMap/ODM #1151: "Cannot call rmtree on a symbolic link" error when running `--split` + `--optimize-disk-space`

tiagodc, 2020-08-24:
**Installation:** Docker Image built on my machine with ODM v1.0.2.

**OS:** Ubuntu 18.04 and I'm using the CLI, so browser info is not relevant.

**Problem:** processing killed when running a large image list with `--optimize-disk-space`. After removing this flag and re-running the same command it worked. The error seems related to `--optimize-disk-space` and `split-merge` being used together, as the `splitmerge.py` file is the only one that creates _symlinks_ (`splitmerge.py` >> lines 123 to 131).

**Error log:**
```
[INFO]    Finished dataset stage                                                                                                                                        
[INFO]    Running split stage                                                                                                                                           
[INFO]    Normal dataset, will process all at once.                                                                                                                     
[INFO]    Finished split stage                                                                                                                                          
[INFO]    Running merge stage                                                                                                                                           
[INFO]    Normal dataset, nothing to merge.                                                                                                                             
[INFO]    Finished merge stage                                                                                                                                          
[INFO]    Running opensfm stage                                                                                                                                         
[WARNING] /datasets/project/submodels/submodel_0000/opensfm/image_list.txt already exists, not rerunning OpenSfM setup                                                  
[WARNING] Detect features already done: /datasets/project/submodels/submodel_0000/opensfm/features exists                                                               
[WARNING] Match features already done: /datasets/project/submodels/submodel_0000/opensfm/matches exists                                                                 
[WARNING] Found a valid OpenSfM tracks file in: /datasets/project/submodels/submodel_0000/opensfm/tracks.csv                                                            
[WARNING] Found a valid OpenSfM reconstruction file in: /datasets/project/submodels/submodel_0000/opensfm/reconstruction.json                                           
[INFO]    Already extracted cameras   Traceback (most recent call last):                                                                                                                                      
  File "/code/run.py", line 62, in <module>                                                                                                                             
    app.execute()                                                                                                                                                       
  File "/code/stages/odm_app.py", line 95, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 350, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 350, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 350, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 331, in run
    self.process(self.args, outputs)
  File "/code/stages/run_opensfm.py", line 37, in process
    shutil.rmtree(octx.path("features"))
  File "/usr/lib/python2.7/shutil.py", line 232, in rmtree
    onerror(os.path.islink, path, sys.exc_info())
  File "/usr/lib/python2.7/shutil.py", line 230, in rmtree
    raise OSError("Cannot call rmtree on a symbolic link")
OSError: Cannot call rmtree on a symbolic link
Traceback (most recent call last):
  File "/code/run.py", line 62, in <module>
    app.execute()
  File "/code/stages/odm_app.py", line 95, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 350, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 331, in run
    self.process(self.args, outputs)
  File "/code/stages/splitmerge.py", line 219, in process
    system.run(" ".join(map(quote, argv)), env_vars=os.environ.copy())
  File "/code/opendm/system.py", line 76, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

**Code:**
```
docker run -ti --rm -v "$(pwd)":/datasets odm:v1.0.2 --project-path /datasets project --dsm --orthophoto-resolution 15 --radiometric-calibration camera --dem-resolution 15 --split 200 --split-overlap 50 --pc-las  --use-hybrid-bundle-adjustment --max-concurrency 3 --use-opensfm-dense --optimize-disk-space
```

closed 2020-08-24 by pierotofy

smathermather, 2020-08-25:
Excellent catch, and thanks for the patch.

tiagodc, 2020-08-25:
you're welcome... I discovered ODM just recently and it's an amazing tool! I'm glad to help :)

commit 8489cf06 (2020-08-24): Merge pull request #1152 from tiagodc/master

fix #1151 - check for symbolic links before removing directories in run_opensfm.py
commit 85ea7c39 (2020-08-24): fix #1151 - check for symbolic links before removing directories in run_opensfm.py