OpenDroneMap/ODM #1154: Different folder for locally run ClusterODM and remotely run NodeODM

jchen706, 2020-08-26:
Dataset used is Phalaborwa. The 36 images. 

I used the ClusterODM locally on Virtualbox ubuntu 16.04. NodeODM was run on another machine. I used the docker configurations. 

The command ran locally was **sudo docker run -ti --rm -v /$(pwd):/datasets/code opendronemap/odm --project-path /datasets --split 18 --split-overlap 75 --use-3dmesh --sm-cluster http://172.17.0.1:4000**

The nodeODM command ran froom the other machine was just **sudo docker run -p 3000:3000 opendronemap/nodeodm.**

I want to get the same folders output from the sub Model 0 ran locally and the sub Model 1. 
Sub Model 1 is missing mve, odm_filterpoints, odm_meshing, odm_texturing ..etc. As you can see in the images below.

Sub Model 0 ran locally
;
[image]

Sub Model 1 ran remotely.
[image]

pierotofy, 2020-08-26:
This sounds like a question? Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

closed 2020-08-26 by pierotofy

smathermather, 2020-08-26:
https://community.opendronemap.org/t/seems-when-a-submodel-is-ran-remotely-the-mve-folder-and-odm-mesh-are-not-generated-compare-to-a-locally-run-split-merge/5231/2