OpenDroneMap/ODM #1159: Scope of Work - Ubuntu 20.04LTS (Core20 SNAP) and WSL Image

Saijin-Naib, 2020-09-16:
### Scope of Work
Update OpenDroneMap and associated blocking libraries to be compatible with Ubuntu 20.04LTS, and Core20 snap packaging.
In addition, from the updated libraries, generate a WSL 1/2 compatible image of the same.

Further, updates to documentation to support any manual changes needed to ensure library compatibility with Ubuntu 20.04LTS and Core20 snap will be generated.

### Timeframe
Projected to be approximately 4 weeks

### Work-Hours
Projected to be approximately 160 hours

### Contractor
Dan Llewellyn

pierotofy, 2020-10-18:
Funding goal achieved! Updates coming soon.

pierotofy, 2020-10-21:
Progress updates will be posted here: https://community.opendronemap.org/t/progress-on-snap-packaging-project-for-odm/5711

pierotofy, 2020-11-06:
I think with https://github.com/OpenDroneMap/ODM/pull/1189 this issue can now be closed.

In the next few days we will do some testing to make sure things are working properly, but otherwise this funding proposal has been completed successfully!

Thanks to all, especially @Saijin-Naib for spearheading the proposal and @diddledan for the implementation! :pray:

closed 2020-11-06 by pierotofy