OpenDroneMap/ODM #1163: ODM crashes due to XMP value after upgrading to 2.0

pqvst, 2020-10-12:
### How did you install OpenDroneMap? (Docker, natively, ...)?

docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 85 on macOS (Catalina)

### What is the problem?

After upgrading to ODM 2.0 I get the following errors:
- An error occurs while parsing XMP data (could not convert string to float: '610/1000')
- Another error occurs while processing this error (object has no attribute 'message')

```
[INFO]    Loading 4 images
Traceback (most recent call last):
  File "/code/opendm/photo.py", line 225, in parse_exif_values
    ], float)
  File "/code/opendm/photo.py", line 256, in set_attr_from_xmp_tag
    setattr(self, attr, cast(v))
ValueError: could not convert string to float: '610/1000'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/code/run.py", line 68, in <module>
    app.execute()
  File "/code/stages/odm_app.py", line 95, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 337, in run
    self.process(self.args, outputs)
  File "/code/stages/dataset.py", line 109, in process
    p = types.ODM_Photo(f)
  File "/code/opendm/photo.py", line 70, in __init__
    self.parse_exif_values(path_file)
  File "/code/opendm/photo.py", line 236, in parse_exif_values
    log.ODM_WARNING("Cannot read XMP tags for %s: %s" % (_path_file, e.message))
AttributeError: 'ValueError' object has no attribute 'message'
```

I traced down the "610/1000" value to the following XMP value in my image:

```
"GPSXYAccuracy": "610/1000"
```

The exact same image (same metadata) worked fine in earlier ODM versions.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

I expect ODM to not crash due to this XMP value.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Add an image that includes the above metadata. I can share a copy of the image privately if needed.

pqvst, 2020-10-12:
After checking the output with ODM 1.0.2 I can see that it logs the following warning:

```
[WARNING] Cannot read XMP tags for /datasets/project/images/foo.jpg: invalid literal for float(): 610/1000
```

So it seems the main issue is the exception that occurs while handling the XMP exception.

pierotofy, 2020-10-12:
Thanks for the report!

Could you share with us such an image?

pqvst, 2020-10-12:
> Could you share with us such an image?

I'd rather not share the image publicly. I've emailed the image to your git email address. I hope you don't mind.

pierotofy, 2020-10-12:
That works, thanks.

PR #1165 (2020-10-14): Add support for XMP fraction values, fix #1163
Fixes #1163
commit 5d3ec490 (2020-10-14): Add support for XMP fraction values, fix #1163
commit 929f1db5 (2020-10-14): Merge pull request #1165 from pierotofy/xmpfix

Add support for XMP fraction values, fix #1163
closed 2020-10-14 by pierotofy