OpenDroneMap/ODM #1180: Snapcraft build fails

pierotofy, 2020-10-30:
Not sure if this is still due to work being in progress, but attempting a build from snapcraft.io leads to some errors.

```
https://launchpad.net/~build.snapcraft.io/+snap/ec5462969ff0cead31e46632aedae124/+build/1167276
RUN: /usr/share/launchpad-buildd/bin/builder-prep 
Kernel version: Linux lgw01-amd64-022 4.15.0-118-generic #119-Ubuntu SMP Tue Sep 8 12:30:01 UTC 2020 x86_64
Buildd toolchain package versions: launchpad-buildd_193~468~ubuntu18.04.1 python3-lpbuildd_193~468~ubuntu18.04.1 sbuild_0.75.0-1ubuntu1 bzr-builder_0.7.3+bzr174~ppa13~ubuntu16.04.1 bzr_2.7.0+bzr6622-10 git-build-recipe_0.3.6~git201906051340.ff11471~ubuntu18.04.1 git_1:2.17.1-1ubuntu0.7 dpkg-dev_1.19.0.5ubuntu2.3 python-debian_0.1.32 python3-debian_0.1.32.
Syncing the system clock with the buildd NTP service...
30 Oct 19:05:31 ntpdate[1799]: adjust time server 10.211.37.1 offset -0.000684 sec
RUN: /usr/share/launchpad-buildd/bin/in-target unpack-chroot --backend=lxd --series=bionic --arch=amd64 SNAPBUILD-1167276 --image-type lxd /home/buildd/filecache-default/41b151d1548e4df3aa6a53cafa07224eb18722e7
Creating target for build SNAPBUILD-1167276
To start your first container, try: lxc launch ubuntu:18.04

RUN: /usr/share/launchpad-buildd/bin/in-target mount-chroot --backend=lxd --series=bionic --arch=amd64 SNAPBUILD-1167276
Starting target for build SNAPBUILD-1167276
/usr/lib/python3/dist-packages/pylxd/models/_model.py:116: UserWarning: Attempted to set unknown attribute "location" on instance of "Container"
  key, self.__class__.__name__
/usr/lib/python3/dist-packages/pylxd/models/_model.py:116: UserWarning: Attempted to set unknown attribute "location" on instance of "Container"
  key, self.__class__.__name__
RUN: /usr/share/launchpad-buildd/bin/in-target override-sources-list --backend=lxd --series=bionic --arch=amd64 SNAPBUILD-1167276 'deb http://ftpmaster.internal/ubuntu bionic main restricted universe multiverse' 'deb http://ftpmaster.internal/ubuntu bionic-security main restricted universe multiverse' 'deb http://ftpmaster.internal/ubuntu bionic-updates main restricted universe multiverse'
Overriding sources.list in build-SNAPBUILD-1167276
/usr/lib/python3/dist-packages/pylxd/models/_model.py:116: UserWarning: Attempted to set unknown attribute "location" on instance of "Container"
  key, self.__class__.__name__
RUN: /usr/share/launchpad-buildd/bin/in-target update-debian-chroot --backend=lxd --series=bionic --arch=amd64 SNAPBUILD-1167276
Updating target for build SNAPBUILD-1167276
Hit:1 http://ftpmaster.internal/ubuntu bionic InRelease
Get:2 http://ftpmaster.internal/ubuntu bionic-security InRelease [88.7 kB]
Get:3 http://ftpmaster.internal/ubuntu bionic-updates InRelease [88.7 kB]
Get:4 http://ftpmaster.internal/ubuntu bionic-security/main amd64 Packages [1399 kB]
Get:5 http://ftpmaster.internal/ubuntu bionic-security/main Translation-en [276 kB]
Get:6 http://ftpmaster.internal/ubuntu bionic-security/restricted amd64 Packages [166 kB]
Get:7 http://ftpmaster.internal/ubuntu bionic-security/restricted Translation-en [22.1 kB]
Get:8 http://ftpmaster.internal/ubuntu bionic-security/universe amd64 Packages [1079 kB]
Get:9 http://ftpmaster.internal/ubuntu bionic-security/universe Translation-en [241 kB]
Get:10 http://ftpmaster.internal/ubuntu bionic-security/multiverse amd64 Packages [12.9 kB]
Get:11 http://ftpmaster.internal/ubuntu bionic-security/multiverse Translation-en [2964 B]
Get:12 http://ftpmaster.internal/ubuntu bionic-updates/main amd64 Packages [1726 kB]
Get:13 http://ftpmaster.internal/ubuntu bionic-updates/main Translation-en [367 kB]
Get:14 http://ftpmaster.internal/ubuntu bionic-updates/restricted amd64 Packages [184 kB]
Get:15 http://ftpmaster.internal/ubuntu bionic-updates/restricted Translation-en [24.6 kB]
Get:16 http://ftpmaster.internal/ubuntu bionic-updates/universe amd64 Packages [1683 kB]
Get:17 http://ftpmaster.internal/ubuntu bionic-updates/universe Translation-en [354 kB]
Get:18 http://ftpmaster.internal/ubuntu bionic-updates/multiverse amd64 Packages [31.9 kB]
Get:19 http://ftpmaster.internal/ubuntu bionic-updates/multiverse Translation-en [6980 B]
Fetched 7753 kB in 8s (1017 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  advancecomp apt base-files bash binutils binutils-common
  binutils-x86-64-linux-gnu bsdutils bzip2 ca-certificates cpp cpp-7 debconf
  dpkg dpkg-dev e2fsprogs fdisk g++ g++-7 gcc gcc-7 gcc-7-base gcc-8-base gpg
  gpg-agent gpgconf gpgv grep libapparmor1 libapt-pkg5.0 libasan4 libatomic1
  libbinutils libblkid1 libbz2-1.0 libc-bin libc-dev-bin libc6 libc6-dev
  libcc1-0 libcilkrts5 libcom-err2 libcryptsetup12 libdb5.3 libdevmapper1.02.1
  libdpkg-perl libext2fs2 libfdisk1 libgcc-7-dev libgcc1 libgcrypt20
  libgnutls30 libgomp1 libidn11 libidn2-0 libitm1 libjson-c3 libkmod2 liblsan0
  libmount1 libmpx2 libncurses5 libncursesw5 libpam-modules libpam-modules-bin
  libpam-runtime libpam0g libperl5.26 libpng16-16 libprocps6 libquadmath0
  libseccomp2 libsmartcols1 libsqlite3-0 libss2 libssl1.1 libstdc++-7-dev
  libstdc++6 libsystemd0 libtinfo5 libtsan0 libubsan0 libudev1 libunistring2
  libuuid1 libzstd1 linux-libc-dev login mount ncurses-base ncurses-bin
  openssl passwd patch perl perl-base perl-modules-5.26 pkgbinarymangler
  procps systemd systemd-sysv tar tzdata ubuntu-keyring util-linux
105 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 70.8 MB of archives.
After this operation, 5928 kB of additional disk space will be used.
Get:1 http://ftpmaster.internal/ubuntu bionic-security/main amd64 libc6-dev amd64 2.27-3ubuntu1.2 [2585 kB]
[... 824 lines cut ...]
snapcraft 4.3 from Canonical* installed
/usr/lib/python3/dist-packages/pylxd/models/_model.py:116: UserWarning: Attempted to set unknown attribute "location" on instance of "Container"
  key, self.__class__.__name__
Running repo phase...
Cloning into 'opendronemap'...
[30/Oct/2020:19:08:31 +0000] "CONNECT github.com:443 HTTP/1.1" 200 11273261 "-" "git/2.17.1"
Running pull phase...
*EXPERIMENTAL* 'package-repositories' configured, but not enabled. Enable with '--enable-experimental-package-repositories' flag.
Build failed
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/lpbuildd/target/build_snap.py", line 257, in run
    self.pull()
  File "/usr/lib/python3/dist-packages/lpbuildd/target/build_snap.py", line 223, in pull
    env=env)
  File "/usr/lib/python3/dist-packages/lpbuildd/target/build_snap.py", line 100, in run_build_command
    return self.backend.run(args, env=full_env, **kwargs)
  File "/usr/lib/python3/dist-packages/lpbuildd/target/lxd.py", line 537, in run
    subprocess.check_call(cmd, **kwargs)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['lxc', 'exec', 'lp-bionic-amd64', '--env', 'LANG=C.UTF-8', '--env', 'SHELL=/bin/sh', '--env', 'SNAPCRAFT_LOCAL_SOURCES=1', '--env', 'SNAPCRAFT_SETUP_CORE=1', '--env', 'SNAPCRAFT_BUILD_INFO=1', '--env', 'SNAPCRAFT_IMAGE_INFO={"build-request-id": "lp-60113921", "build-request-timestamp": "2020-10-30T19:04:52Z", "build_url": "https://launchpad.net/~build.snapcraft.io/+snap/ec5462969ff0cead31e46632aedae124/+build/1167276"}', '--env', 'SNAPCRAFT_BUILD_ENVIRONMENT=host', '--env', 'http_proxy=http://10.10.10.1:8222/', '--env', 'https_proxy=http://10.10.10.1:8222/', '--env', 'GIT_PROXY_COMMAND=/usr/local/bin/snap-git-proxy', '--', '/bin/sh', '-c', 'cd /build/opendronemap && linux64 snapcraft pull']' returned non-zero exit status 2.
Revoking proxy token...
RUN: /usr/share/launchpad-buildd/bin/in-target scan-for-processes --backend=lxd --series=bionic --arch=amd64 SNAPBUILD-1167276
Scanning for processes to kill in build SNAPBUILD-1167276
```

pierotofy, 2020-10-30:
It probably has to to with the `--enable-experimental-package-repositories` flag. So maybe a config issue in the snapcraft store.

Saijin-Naib, 2020-10-30:
Dan made me this magic yml file that works, but his custom build Actions to generate x86 aren't working ATM, so this is x64 only.

https://github.com/Saijin-Naib/ODM/blob/master/.github/workflows/snapcraftbuild.yml

pierotofy, 2020-10-30:
Could you open a pull request for it?

Saijin-Naib, 2020-10-30:
PR #1181 should take care of it.

The SECRET is for my account, so a new one will need to be generated to upload to the proper SNAP store.

closed 2020-11-02 by pierotofy