OpenDroneMap/ODM #1196: Native Build Failing

smathermather, 2020-11-07:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Natively

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Ubuntu 20.04

### What is the problem?

Build failing

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Build not failing

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Build on 20.04, possibly getting the following error:
```
/home/gisuser/ODM/modules/odm_georef/src/Georef.cpp: In member function ‘bool Georef::readHeader(const string&, pcl::PCLPointCloud2&, Eigen::Vector4f&, Eigen::Quaternionf&, int&, int&, unsigned int&, int)’:
/home/gisuser/ODM/modules/odm_georef/src/Georef.cpp:1822:24: warning: conversion from ‘std::size_t’ {aka ‘long unsigned int’} to ‘uint32_t’ {aka ‘unsigned int’} may change value [-Wconversion]
 1822 |     cloud.width      = nr_point;
      |                        ^~~~~~~~
[ 92%] Linking CXX executable ../../bin/odm_orthophoto
[ 92%] Built target odm_orthophoto
[100%] Linking CXX executable ../../bin/odm_georef
/usr/bin/ld: CMakeFiles/odm_georef.dir/src/Georef.cpp.o: undefined reference to symbol 'pthread_condattr_setclock@@GLIBC_2.3.3'
/usr/bin/ld: /lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [modules/odm_georef/CMakeFiles/odm_georef.dir/build.make:421: bin/odm_georef] Error 1
make[1]: *** [CMakeFiles/Makefile2:115: modules/odm_georef/CMakeFiles/odm_georef.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

pierotofy, 2020-11-07:
Does the Docker image build OK (which is based on 20.04 also?)

smathermather, 2020-11-07:
Yup. Docker build works fine. I know it's a fools errand to build native, but I was trying to isolate the issues I'm seeing in split merge, so a native build seemed like a good sanity check.

pierotofy, 2020-11-07:
Ah, interesting; I wonder what are the difference in 20.04 from a native install and the docker environment.

pierotofy, 2020-12-06:
Is this still an issue?

smathermather, 2020-12-10:
I'll be re-testing tomorrow.

smathermather, 2020-12-11:
It now builds and runs fine natively.

closed 2020-12-11 by smathermather