OpenDroneMap/ODM #1227: Add PDF reports while using split-merge

pierotofy, 2021-01-15:
Currently ODM does not generate PDF reports when using split-merge. It could be possible to generate them by:

- Recomputing/aggregating stats from submodels
- Write a new report generation code (separate from OpenSfM's)

smathermather, 2021-01-15:
I'd lean toward aggregate stats. Any reason to not have submodel stats as well?:for larger datasets it could be more useful than global stats.

pierotofy, 2021-01-15:
I agree to both. The bottom line is that the split-merge report would contain slightly different information compared to normal runs.

smathermather, 2021-01-15:
Yup. Agreed.

YanNoun, 2021-01-18:
@pierotofy `opensfm/action/compute_statistics.py` calls `stats.py` functions that works on a list of reconstructions.

pierotofy, 2021-01-18:
Didn't know that. Thanks @YanNoun ! I'll try it out.

YanNoun, 2021-01-18:
The `save_heatmap`/`save_topview` might be super slow with 100K/1M points, feel free to run some random index-based decimation.

pierotofy, 2021-01-18:
I did notice some slowdowns with larger datasets; would this be a good idea for a PR to OpenSfM for the `compute_statistics` command? (I could add a `--decimate` flag)

YanNoun, 2021-01-18:
Yup definitely, that would be great !

pierotofy, 2021-01-18:
https://github.com/mapillary/OpenSfM/pull/695