OpenDroneMap/ODM #1237: QGIS 3.18 and ODM-generated EPTs

Saijin-Naib, 2021-03-04:
In testing EPTs generated by ODM, I came across an issue with them displaying properly within QGIS 3.18.x with PDAL support.

I've provided sample data to Saber Razmjooei/Lutra Consulting, and they indicated that the ODM-generated EPT they tested did not have any Z-data, but the accompanying LAZ files did.

Is there anything in our process that might lead to Z-Values being dropped from EPTs during generation?

```
I had a quick look at the OpenDrone data. It looks fine in 2D map canvas. But does not have any Z values and hence no 3D map view is available.

When I open the individual laz files, they do contain Z value and render properly in 3D.

Could you maybe try to use a more recent version of PDAL/Untwine to create your composite lidar and see if it is still the same result in QGIS?

Kind regards
Saber
```

pierotofy, 2021-03-04:
That seems strange; a missing Z-value would imply that the EPT cannot be opened in Potree/WebODM also (but it can).

That said, ODM probably used Entwine (not Untwine) to generate the EPT dataset, we use Untwine only as a fallback if an out-of-memory error happens. The two should produce comparable EPT data though, so it shouldn't matter.

I would start debugging by trying to regenerate the EPT dataset manually (using the version of entwine provided with ODM, then try untwine, then switch to a standalone version of entwine, and finally a standalone version of untwine).

Saijin-Naib, 2021-03-04:
Got this feedback from Martin Dobias on the test EPT from ODM:  
```
- The indexed EPT dataset does not include statistics for each dimension (in the ept.json file, for each dimension there should be
entries like "minimum", "maximum" and few others). Because of that, some functionality in QGIS does not work well. It would be great if OpenDroneMap run the indexing including statistics - it will help the usage in QGIS a lot - can you please ask ODM team to add that? (This is also the reason why e.g. Attribute by ramp rendering does not work properly - we do not have min/max values for dimensions, so QGIS defaults to zeroes and nothing is rendered...
```

pierotofy, 2021-03-04:
Cool, perhaps entwine needs to be updated then.

Saijin-Naib, 2021-03-04:
> Cool, perhaps entwine needs to be updated then.

Looking at Entwine/Untwine, it looks like we're already targeting the latest Entwine, right? And we're using your fork of Untwine, as well. Does that contain changes that the upstream Hobu Untwine doesn't have?

smathermather, 2021-03-04:
> I'll try to tackle this in the coming week or so.

I took a look whether there are flags for generating stats, and there's nothing: a strong indication that it generates them if a new enough version is used. Glad you'll be working on this, and thanks for troubleshooting with Lutra: super cool to see this in Q... .

I wonder if it will fix this issue:
https://community.opendronemap.org/t/entwine-point-cloud-vs-laz-and-everything-else/6874

pierotofy, 2021-03-04:
For whatever reason the EPT I generated on my machine for brighton works OK.

[image]

[ept.json.txt](https://github.com/OpenDroneMap/ODM/files/6085307/ept.json.txt)

Saijin-Naib, 2021-03-04:
That's super strange... It is missing the values that Martin said it should have.

pierotofy, 2021-03-04:
And this one is generated with untwine (also from ODM):

[image]

[ept.json.txt](https://github.com/OpenDroneMap/ODM/files/6085331/ept.json.txt)

wonder-sk, 2021-03-04:
Hi,

sorry the "missing Z values" was an incorrect finding - the Z values are present as they should be :-)

The only missing thing, as mentioned in the mail, is that the generated EPT does not include statistics. Not sure if ODM uses Entwine or Untwine (or both?) - but either has an option to include statistics for each dimension. This is extremely helpful for QGIS as we then know what are the valid ranges of dimensions and can pre-configure 2D/3D rendering to look good out of the box...

Saijin-Naib, 2021-03-04:
> Not sure if ODM uses Entwine or Untwine (or both?)

Piero has stated that it uses Entwine by default unless we get an out of memory situation, in which case it'll switch to Untwine.

Do you happen to have a link to specs or an example with those properly formed JSON files containing all the extra stats for proper QGIS display?

pierotofy, 2021-03-04:
@Saijin-Naib can you upload your ept.json somewhere also?

Saijin-Naib, 2021-03-04:
This is the EPT.JSON that was in the Living Room sample data that Martin/Saber were poking at.
[ept.json.txt](https://github.com/OpenDroneMap/ODM/files/6085876/ept.json.txt)

wonder-sk, 2021-03-04:
> Do you happen to have a link to specs or an example with those properly formed JSON files containing all the extra stats for proper QGIS display?

Here's an example with and without stats in ept.json for the same source file:
- [ept.json.no-stats.txt](https://github.com/OpenDroneMap/ODM/files/6085858/ept.json.no-stats.txt)
- [ept.json.with-stats.txt](https://github.com/OpenDroneMap/ODM/files/6085859/ept.json.with-stats.txt)

Unfortunately the statistics are not described in the official EPT docs: https://entwine.io/entwine-point-tile.html
I had a discussion with entwine/untwine devs and the idea is to keep that outside of "core" spec - and maybe introduce a simple system of extensions (where statistics would be one such extension)

smathermather, 2021-03-04:
> For whatever reason the EPT I generated on my machine for brighton works OK. (From ODM using --pc-ept).

Nope: those don't look like they are in the same place. Your error is just smaller than mine maybe because I have a whole city of point cloud and you have a picnic area. :smiley_cat:

pierotofy, 2021-03-04:
Georeferencing offset problems aside, it's strange my EPT dataset displays correctly, because it doesn't have statistics either.

Full dataset included here: 

[entwine_pointcloud.zip](https://github.com/OpenDroneMap/ODM/files/6086320/entwine_pointcloud.zip)

wonder-sk, 2021-03-04:
> Georeferencing offset problems aside, it's strange my EPT dataset displays on the map, because it doesn't have statistics either.

QGIS is able to load EPT datasets also when statistics are missing, the user experience is just not as smooth.

In QGIS we use those statistics mainly to know min-max ranges for Z and other dimensions - so for example in your dataset when I wanted to use "Attribute by ramp" rendering style based on Z, it took me a some trial and error that the usable range is somewhere around 150-180:

[image]

 With point clouds with RGB this shortcoming is maybe less obvious than with non-colorized lidar data where we can't default to RGB renderer.

Maybe we will add also a "load min/max from the current map view" button to estimate the ranges - but we don't have that yet...

pierotofy, 2021-03-05:
That makes sense :) Thanks for the clarification @wonder-sk !

smathermather, 2021-03-05:
If it's useful, I have a dataset that if I try to access properties in QGIS crashes it every time (on PopOS anyway).

wonder-sk, 2021-03-05:
@smathermather I think the crash when opening layer properties should be already fixed - see https://github.com/qgis/QGIS/issues/41722 - hopefully we will get 3.18.1 out soon with this and some other fixes. Or you can compile QGIS from git master for the time being...

smathermather, 2021-03-05:
Ah, perfect. Glad it was already found.

Saijin-Naib, 2021-03-11:
I have another LAS/LAZ dataset that doesn't generate a proper EPT in QGIS when being previewed:  
https://community.opendronemap.org/t/help-entwine-at-the-beginning/6979

wonder-sk, 2021-03-12:
> I have another LAS/LAZ dataset that doesn't generate a proper EPT in QGIS when being previewed:
> https://community.opendronemap.org/t/help-entwine-at-the-beginning/6979

@Saijin-Naib I have just tried it and this is a known issue in untwine when scale of X/Y/Z coordinates is ridiculously low - see https://github.com/hobu/untwine/issues/53#issuecomment-797381755
If ODM produces files with such scaling, maybe it would be good to force default precision of coordinates to e.g. 1mm

Saijin-Naib, 2021-03-12:
@wonder-sk ,

These data are apparently directly out of a Faro S70 Laser Scanning Unit, a fairly high-end piece of kit.

pierotofy, 2021-06-14:
Closing, as this doesn't seem to be a bug in ODM per-se, just a data input workflow.

closed 2021-06-14 by pierotofy