OpenDroneMap/ODM #1238: Support of 8bit multi band JPEG + 16bit single band TIFF images

zgyivi, 2021-03-04:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker as part of WebODM

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 86 on Ubuntu Linux

### What is the problem?

DJI Multispectral camera provides 8bit RGB JPG files with EXIF data + 5 separate 16bit single band TIFF images
When I try to process such a set OpenMVS/DensifyPointCloud reports error, because the undistorted RGB images
are converted to 16bit 3band TIFF format and it cannot process that.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Undistorting should keep 8bit format for multiband images even if there are 16bit single band files in the set.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

The problem occur on the image set which was shared by mapninja on community page
[DJI Phantom 4 Pro Multipectral w/o RTK](https://community.opendronemap.org/t/dji-phantom-4-pro-multipectral-w-o-rtk/4978?u=zgyivi)
Only the JPGs can be processed properly, but when all images added processing fails as described above.
Consol log shows ([link to whole log](https://drive.google.com/file/d/1RdW8SGt7cC2HfRbV_bH9DV8ZPDp9ECpY/view?usp=sharing)): 
`[INFO]    Initializing ODM - Sat Feb 27 23:14:17  2021
[INFO]    ==============
[INFO]    build_overviews: False
[INFO]    camera_lens: auto
[INFO]    cameras: {}
[INFO]    crop: 3
[INFO]    debug: False
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 5
[INFO]    depthmap_resolution: 640
[INFO]    dsm: False
[INFO]    dtm: False
[INFO]    end_with: odm_report
[INFO]    fast_orthophoto: False
[INFO]    feature_quality: high
[INFO]    feature_type: sift
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    geo: None
[INFO]    gps_accuracy: 10
[INFO]    ignore_gsd: False
[INFO]    matcher_distance: 0
[INFO]    matcher_neighbors: 8
[INFO]    matcher_type: flann
[INFO]    max_concurrency: 16
[INFO]    merge: all
[INFO]    mesh_octree_depth: 11
[INFO]    mesh_size: 200000
[INFO]    min_num_features: 8000
[INFO]    name: 9357efe3-2a90-4fc2-b82c-be94813d33fb
[INFO]    optimize_disk_space: False
[INFO]    orthophoto_compression: DEFLATE
[INFO]    orthophoto_cutline: False
[INFO]    orthophoto_no_tiled: False
[INFO]    orthophoto_png: False
[INFO]    orthophoto_resolution: 5
[INFO]    pc_classify: False
[INFO]    pc_csv: False
[INFO]    pc_ept: False
[INFO]    pc_filter: 2.5
[INFO]    pc_las: False
[INFO]    pc_quality: medium
[INFO]    pc_rectify: False
[INFO]    pc_sample: 0
[... 179 lines cut ...]
23:24:28 [App     ] OS: Linux 4.15.0-134-generic (x86_64)
23:24:28 [App     ] SSE & AVX compatible CPU & OS detected
23:24:28 [App     ] Command line: /var/www/data/9357efe3-2a90-4fc2-b82c-be94813d33fb/opensfm/undistorted/openmvs/scene.mvs --resolution-level 2 --min-resolution 320 --max-resolution 1397 --max-threads 16 --number-views-fuse 2 -w /var/www/data/9357efe3-2a90-4fc2-b82c-be94813d33fb/opensfm/undistorted/openmvs/depthmaps -v 0
23:24:28 [App     ] error: failed loading image header
23:24:28 [App     ] error: failed reloading image '/var/www/data/9357efe3-2a90-4fc2-b82c-be94813d33fb/opensfm/undistorted/images/DJI_0280.JPG.tif'
23:24:28 [App     ] error: invalid project
Traceback (most recent call last):
File "/code/run.py", line 68, in <module>
app.execute()
File "/code/stages/odm_app.py", line 81, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 338, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 338, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 338, in run
self.next_stage.run(outputs)
[Previous line repeated 1 more time]
File "/code/opendm/types.py", line 319, in run
self.process(self.args, outputs)
File "/code/stages/openmvs.py", line 60, in process
system.run('%s "%s" %s' % (context.omvs_densify_path,
File "/code/opendm/system.py", line 79, in run
raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1`

huazai-1994, 2022-01-24:
My multispectral images were acquired by DJI drones ，and the images contain 8bit RGB JPG files with EXIF data + 5 separate 16bit single band TIFF images. I just put all the images in a folder named "images", and I can get a  result  without running error.But there is something wrong with the result,which exists severely missing
[image]
.

pierotofy, 2022-07-26:
This is fixed with #1512 , thanks for sharing an input dataset to test with! :pray:

closed 2022-07-26 by pierotofy