OpenDroneMap/ODM #1240: qemu: uncaught target signal 4 (Illegal instruction) - core dumped

shawndearmond, 2021-03-08:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker Desktop preview 3.1.0

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

MacBook Pro M1 (Apple Silicon)
macOS 11.2.2 (Big Sur)

### What is the problem?

Following the directions here: https://github.com/OpenDroneMap/ODM#quickstart

I run:
`docker run -ti --rm -v /path/to/datasets:/datasets opendronemap/odm --project-path /datasets project`
And I get:
`WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
qemu: uncaught target signal 4 (Illegal instruction) - core dumped`
And it just sits there, never bringing me back to a cli prompt.

I also tried:
`docker run --platform linux/amd64 -ti --rm -v /path/to/datasets:/datasets opendronemap/odm --project-path /datasets project` (note: specify `--platform linux/amd64`)
And I get:
`qemu: uncaught target signal 4 (Illegal instruction) - core dumped`
It didn't complain about the image's platform this time, so that's better! Would be nice if I could use the arm64 image, though.
And still no prompt.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Well, this is the first time I've tried running ODM, so I'm not quite sure. 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Use Docker Desktop preview 3.1.0 on an M1 Mac
git clone ODM from Github
cd ODM
Run the commands above.

Possible related issues:
https://github.com/docker/for-mac/issues/5251
https://github.com/multiarch/qemu-user-static/issues/99

shawndearmond, 2021-03-08:
This line from [Docker's M1 page](https://docs.docker.com/docker-for-mac/apple-m1/) is likely relevant too: 

`However, attempts to run Intel-based containers on Apple M1 machines can crash as QEMU sometimes fails to run the container. Therefore, we recommend that you run ARM64 containers on M1 machines. These containers are also faster and use less memory than Intel-based containers.`

pierotofy, 2021-03-08:
ODM does not support M1 chips at the moment. It's something we'd like to add at some point in the future.

fran2a, 2021-04-14:
Any news on this request? Thanks

pierotofy, 2021-04-14:
Still not supported.

charliep, 2021-05-23:
I have been able to build an ODM image on an M1 chipset but when I run it generates a missing python module error message:

Traceback (most recent call last):
  File "/code/run.py", line 10, in <module>
    from opendm import config
  File "/code/opendm/config.py", line 6, in <module>
    from appsettings import SettingsParser
ModuleNotFoundError: No module named 'appsettings'

The script I hacked to build the image is:
docker system prune --volumes -f
rm -rf github/
mkdir github
cd github
git init
git clone https://github.com/OpenDroneMap/OpenDroneMap.git
cd OpenDroneMap/
docker build --platform linux/arm64/v8 -t odm_m1 --no-cache .

I am a complete newbie to building Docker images, is this even the correct approach?

charliep, 2021-07-06:
ping... Could someone answer my question about the correct way to build ODM?

manand881, 2021-07-07:
currently not possible. please add a post in the community forum https://community.opendronemap.org/

pierotofy, 2021-09-09:
We now have M1 support! :clinking_glasses:

closed 2021-09-09 by pierotofy