OpenDroneMap/ODM #1244: coords.txt is in int, but opensfm reference is in lat,lon (float)

mattip, 2021-03-17:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Natively

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Ubuntu 20.04, Firefox

### What is the problem?

When trying to map image pixels from the OpenSFM reconstruction `shots` in `lat`,`lon` to the OpenDroneMap odm_orthophoto.tif, I noticed a discrepancy of ~0.5 meter. I tracked this down to the rounding of the corner of the gdal projection used to create the orthophoto (as can be seen in the second line of the `odm_referencing/coords.txt` vs. the float values in the `odm_georeferencing/odm_georeferencing_model_geo.txt`). It seems my mistake was compounded by something in the use of gdal to set the output transform, since
```
gdal.Open(<output>).GetGeoTransform()
```

returns the float values, even though it seems the file is created with the integer values.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Would it be possible to use the float values instead of the integer values across the board?

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

The result is independent of any input parameter I could find.

See also [this discussion](https://github.com/mapillary/OpenSfM/issues/657) at OpenSFM about how to map OpenSFM coordinates.

pierotofy, 2021-03-17:
The `reference_lla.json` is not going to match `odm_referencing/coords.txt` (like you say, it will be off by about ~.5 meters). I just prefer to work with UTM offsets (rounded to the next integer) rather than lat/lon.

Is there a reason you can't use `coords.txt`?

Also note that the `reconstruction.json` file is shifted to the offset in `coords.txt` right after reconstruction. So you can't use `reference_lla.json` after https://github.com/OpenDroneMap/ODM/blob/master/stages/run_opensfm.py#L70 is executed.

We'd welcome a PR to improve this, although I'd like to keep the requirement of keeping the offset in integers.

mattip, 2021-03-17:
> Is there a reason you can't use coords.txt?

I was not using any of the `odm_*` files at all. I read the lat,lon reference from `opensfm/reconstruction.json` via OpenSFM's utilities and used `gdal` routines to read the orthophoto ( `gdal.Open(<output>).GetGeoTransform()`). 

> update `reference_lla.json` back with the new offset

I see that has the same value as the `reference` field of the `reconstruction.json`. At what point are they synchronized?

I stopped at the line you pointed to (nice dev setup process for such a complicated piece of software BTW) in a debugger and the `reconstruction.georef.utm_east_offset`, `reconstruction.georef.utm_north_offset` are already rounded to the nearest int at that point.

pierotofy, 2021-03-18:
> At what point are they synchronized?

Good question! I'm not really sure :)

pierotofy, 2021-08-03:
This can be closed (fixed).

closed 2021-08-03 by pierotofy