OpenDroneMap/ODM #1250: Altitude below sea level

originlake, 2021-03-26:
This issue has been there for a while that opensfm doesn't handle the EXIF data "GPS Altitude Ref" correctly. If all the images' GPS Altitude is below sea level, then the model will be upside down. If half of the images have above sea level altitude, the other half have below sea level altitude, it could be even worse. I mentioned this issue in https://github.com/mapillary/OpenSfM/issues/647 and provided a fix to it earlier. 
I don't know the plan of merging opensfm, but would be glad to see it merged in ODM.

pierotofy, 2021-03-26:
We'll be updating OpenSfM in the upcoming weeks (hopefully), so that fix should make it into ODM then.

If you need it sooner, open a PR with the cherry-picked commits in our fork and target the `246` branch? :pray:

originlake, 2021-03-26:
If it's in the plan, I'll wait then. Thanks

closed 2021-03-26 by originlake