OpenDroneMap/ODM #1252: ValueError: Unknown feature type (must be SURF, SIFT, AKAZE, HAHOG or ORB)

Saijin-Naib, 2021-03-29:
****************************************

### How did you install OpenDroneMap? (Docker, natively, ...)?
WSL2, latest ../code directory, latest libraries (apt update), and python3-pyopencl installed. Running under Windows 10 Dev Fast host with CUDA WSL2 NVIDIA Drivers.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
N/A

### What is the problem?
ValueError: Unknown feature type (must be SURF, SIFT, AKAZE, HAHOG or ORB)

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
Features passed to opensfm are accepted properly.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Private "livingroom" dataset, which normally processes just fine.

```
root@GreenMachine:/code# ./run.sh --rerun-all --crop 0 --use-hybrid-bundle-adjustment
[INFO]    Initializing ODM - Mon Mar 29 16:25:02  2021
[INFO]    ==============
[INFO]    build_overviews: False
[INFO]    camera_lens: auto
[INFO]    cameras: {}
[INFO]    crop: 0.0
[INFO]    debug: False
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 5
[INFO]    depthmap_resolution: 640
[INFO]    dsm: False
[INFO]    dtm: False
[INFO]    end_with: odm_report
[INFO]    fast_orthophoto: False
[INFO]    feature_quality: high
[INFO]    feature_type: sift
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    geo: None
[INFO]    gps_accuracy: 10
[INFO]    ignore_gsd: False
[INFO]    matcher_distance: 0
[INFO]    matcher_neighbors: 8
[INFO]    matcher_type: flann
[INFO]    max_concurrency: 8
[INFO]    merge: all
[INFO]    mesh_octree_depth: 11
[INFO]    mesh_size: 200000
[INFO]    min_num_features: 8000
[INFO]    name: code
[INFO]    optimize_disk_space: False
[INFO]    orthophoto_compression: DEFLATE
[INFO]    orthophoto_cutline: False
[INFO]    orthophoto_no_tiled: False
[INFO]    orthophoto_png: False
[INFO]    orthophoto_resolution: 5
[INFO]    pc_classify: False
[INFO]    pc_csv: False
[INFO]    pc_ept: False
[INFO]    pc_filter: 2.5
[INFO]    pc_las: False
[INFO]    pc_quality: medium
[INFO]    pc_rectify: False
[INFO]    pc_sample: 0
[INFO]    pc_tile: False
[INFO]    primary_band: auto
[INFO]    project_path: /
[INFO]    radiometric_calibration: none
[... 115 lines cut ...]
    raise ValueError(
ValueError: Unknown feature type (must be SURF, SIFT, AKAZE, HAHOG or ORB)
Traceback (most recent call last):
  File "/code/run.py", line 68, in <module>
    app.execute()
  File "/code/stages/odm_app.py", line 82, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 338, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 338, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 338, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 319, in run
    self.process(self.args, outputs)
  File "/code/stages/run_opensfm.py", line 36, in process
    octx.feature_matching(self.rerun())
  File "/code/opendm/osfm.py", line 285, in feature_matching
    self.run('detect_features')
  File "/code/opendm/osfm.py", line 27, in run
    system.run('%s/bin/opensfm %s "%s"' %
  File "/code/opendm/system.py", line 79, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

pierotofy, 2021-03-29:
ODM detects `python3-pyopencl` and assumes you have GPU support. Is this a native install?

Saijin-Naib, 2021-03-29:
> ODM detects `python3-pyopencl` and assumes you have GPU support. Is this a native install? Make sure you have the latest OpenSfM.

WSL2 install, updated via git pull and apt update && apt upgrade, and manual install of python3-pyopenCL. 
W10 Insider Dev host with experimental NVIDIA CUDA WSL2-enabled drivers.
https://developer.nvidia.com/cuda/wsl/download

So in theory, I should have properly working OpenCL.

Output of clinfo:  

```
root@GreenMachine:/code# clinfo
Number of platforms                               1
  Platform Name                                   Portable Computing Language
  Platform Vendor                                 The pocl project
  Platform Version                                OpenCL 1.2 pocl 1.4, None+Asserts, LLVM 9.0.1, RELOC, SLEEF, DISTRO, POCL_DEBUG
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             POCL

  Platform Name                                   Portable Computing Language
Number of devices                                 1
  Device Name                                     pthread-Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz
  Device Vendor                                   GenuineIntel
  Device Vendor ID                                0x6c636f70
  Device Version                                  OpenCL 1.2 pocl HSTR: pthread-x86_64-pc-linux-gnu-skylake
  Driver Version                                  1.4
  Device OpenCL C Version                         OpenCL C 1.2 pocl
  Device Type                                     CPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               8
  Max clock frequency                             4008MHz
  Device Partition                                (core)
    Max number of sub-devices                     8
    Supported partition types                     equally, by counts
    Supported affinity domains                    (n/a)
  Max work item dimensions                        3
  Max work item sizes                             4096x4096x4096
  Max work group size                             4096
  Preferred work group size multiple              8
  Preferred / native vector sizes
    char                                                16 / 16
    short                                               16 / 16
    int                                                  8 / 8
    long                                                 4 / 4
    half                                                 0 / 0        (n/a)
    float                                                8 / 8
    double                                               4 / 4        (cl_khr_fp64)
  Half-precision Floating-point support           (n/a)
  Single-precision Floating-point support         (core)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  Yes
  Double-precision Floating-point support         (cl_khr_fp64)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
  Address bits                                    64, Little-Endian
[... 32 lines cut ...]
  Device Extensions                               cl_khr_byte_addressable_store cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_3d_image_writes cl_khr_fp64 cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_fp64

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Portable Computing Language
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [POCL]
  clCreateContext(NULL, ...) [default]            Success [POCL]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 Portable Computing Language
    Device Name                                   pthread-Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  Success (1)
    Platform Name                                 Portable Computing Language
    Device Name                                   pthread-Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 Portable Computing Language
    Device Name                                   pthread-Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1
```

pierotofy, 2021-03-30:
Make sure you have rebuilt the SuperBuild directory; you need the latest OpenSfM patches to get GPU to work.

Saijin-Naib, 2021-03-30:
Yeah, not native. WSL2 generated by the github actions, but updated code directory by using git pull and updated libs in the install by using apt update/upgrade.

How would I trigger the rebuild of the SuperBuild directory?

Saijin-Naib, 2021-03-31:
> Natively you would do:
> 
> `bash configure.sh reinstall`

Ran that successfully, still failed when python3-pyopencl is present... Maybe I'm missing an OpenCL driver?

pierotofy, 2021-04-16:
Mm, not sure. But if you set `ODM_NO_GPU=YES` in the environment variable list, you can make sure that ODM does not attempt to use the GPU driver. I'm going to close this as it looks like it might be a configuration / build issue. :pray:

closed 2021-04-16 by pierotofy