OpenDroneMap/ODM #1257: WSL import - rootfs file is not in the releases page

tteke, 2021-04-05:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Not applicable (not there yet)

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Windows 10

### What is the problem?

Cannot find rootfs.tar.gz file for wsl import in the releases page as the docs suggest.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Either docs should be changed or the file should be added to releases page

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Not applicable

tteke, 2021-07-09:
I've managed to get a file ready to be imported with wsl but it's around 1.3GB. How do you prefer I share it with you @pierotofy

pierotofy, 2021-07-10:
Ideally we would fix the GitHub workflow file that is in charge on generating the file. But you can share your file via Dropbox/Google Drive and link it to this thread so that we can point people to it as a workaround in the meanwhile!

tteke, 2021-07-11:
heres the drive link https://drive.google.com/file/d/1ncrABdBfuYQ0J89KhsAoVjiT5zac6OKm/view?usp=sharing

pierotofy, 2023-10-29:
Closing, as we no longer recommend to use WSL to run ODM on Windows (we have a native build).

closed 2023-10-29 by pierotofy