OpenDroneMap/ODM #1261: There are many holes in the orthophoto. Is there something wrong with the parameters I set?

githubhanjunjun, 2021-04-07:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What is the problem?

There are many holes in the orthophoto

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

There is a big difference between the orthophoto I generated and the orthophoto I generated in Webodm

My Orthophoto:

[image]

WebODM Orthophoto:

[image]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Options:
[image]

I use nodejs to call ODM's run.sh utility and pass in the parameters shown in the figure above

pierotofy, 2021-04-07:
Are you on Windows? It reminds me of this issue: https://github.com/OpenDroneMap/ODM/issues/872 maybe it's related.

githubhanjunjun, 2021-04-08:
> 你在窗户上吗？这让我想起了这个问题[：#872](https://github.com/OpenDroneMap/ODM/issues/872)也许它是相关的。
No, just the opposite. 
Webodm runs on Windows Docker, and it works very well, but my code runs in Ubuntu, and the effect is very bad

githubhanjunjun, 2021-04-08:
Can you tell me the parameter value that webodm passes when executing ODM's run.sh?

pierotofy, 2021-04-16:
You might be using different versions perhaps? I'm going to close this, as perhaps this is more of a question and not a bug per-se.

Let's move the conversation to the forum if you have questions. :pray:

closed 2021-04-16 by pierotofy