OpenDroneMap/ODM #1267: Feature Request: KMZ output

smathermather, 2021-04-15:
GeoTIFFs are super useful, but KMZ output from ODM would be more widely shareable and portable for a lot of use cases, and gets the data in the hands of non-GIS folks.

geospatialmapmashup, 2021-04-15:
If anything OGC GPKG GeoPackage NOT KMZ. Not knowing anyone that uses that or Google Earth anymore.
GPKG is the standard.

smathermather, 2021-04-15:
I certainly prefer GPKG, but that's a separate feature request. :slightly_smiling_face:

pierotofy, 2021-04-15:
Just a matter of running `gdal_translate odm_orthophoto/odm_orthophoto.tif odm_orthophoto/odm_orthophoto.kmz` and adding a `--orthophoto-kmz` flag. To be added in https://github.com/OpenDroneMap/ODM/blob/master/opendm/orthophoto.py#L58

A PR would be most welcome.

smathermather, 2021-04-15:
https://github.com/OpenDroneMap/ODM/pull/1268

smathermather, 2021-04-15:
Complete (at least when merged).

smathermather, 2021-04-15:
> If anything OGC GPKG GeoPackage NOT KMZ. Not knowing anyone that uses that or Google Earth anymore.
> GPKG is the standard.

@geospatialmapmashup -- if you want to create a separate issue and pull request for GPKG, it should be a pretty quick patch.

closed 2021-04-15 by smathermather