OpenDroneMap/ODM #1272: Feature Request: Write out sidecars for Georeferenceable OBJ

Saijin-Naib, 2021-04-20:
### How did you install OpenDroneMap? (Docker, natively, ...)?
N/A
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
N/A
### What is the problem?
The Wavefront OBJ file format has no concept of CRS and works instead in a local coordinate system (0,0,0). This can present difficulties where people expect to be able to load the model in its proper location.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
Pix4D approaches this limitation of the OBJ format by writing an XYZ offset file in their processing pipeline, along with a PRJ file that defines the projection that the offset applies to. It then is the onus of the ingesting software to take advantage of these offset/spatial reference files to properly locate the OBJ.

https://support.pix4d.com/hc/en-us/articles/204606535-How-to-visualize-the-3D-textured-mesh-with-georeference

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Try to geolocate any OBJ file generated by ODM.

pierotofy, 2021-04-20:
Well, we already do this; the information is in `odm_georeferencing/odm_georeferencing_model_geo.txt`.

Saijin-Naib, 2021-04-20:
> Well, we already do this; the information is in `odm_georeferencing/odm_georeferencing_model_geo.txt`. Are there tools that expect the offset in a particular format matching that of Pix4D?
I thought that's what that file was for...

I'm not entirely certain yet what tools would expect it in that split xyz/prj format. I'm trying to assist a user on the forum in this issue (https://community.opendronemap.org/t/upload-webodm-output-to-arcgis-pro/7383/13?u=saijin_naib) and this is all new to me.

The xyz/prj stuff smacks of ESRI, though. Unfortunately, I lack the license/extensions to do any testing of that workflow.

pierotofy, 2021-04-21:
Ah, I see; OBJ files don't have a (standard) way to specify coordinate systems. If we can identify tools that use a particular convention we could consider adopting it. But until then, it's up to users to specify that information in whatever program they use.

closed 2021-04-26 by pierotofy