OpenDroneMap/ODM #1279: Processing Sentera Double 4K multispectral imagery

amlobat2, 2021-05-03:
### How did you install OpenDroneMap? (Docker, natively, ...)?

I installed WebODM using Git. 

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 90 on Windows 10 

### What is the problem?

The issues along with discussion, datasets, sample EXIF data and output error logs can be found here:

https://community.opendronemap.org/t/sentera-double-4k-multispectral-processing/7473

quick summary: 
I have two 3-band multispectral images, one containing information from the red, green and blue spectral bands (RGB image) and one containing information from the red edge, near IR and a garbage spectral bands (ReGN). I modified the EXIF data for the images and it seems to be triggering the multi-camera setup. However, I get errors each time. It was suggested that it might be a bug in the `DensifyPointCloud` which cannot for some reason read the images generated during the pipeline.

pierotofy, 2022-07-26:
Hey @amlobat2 :hand: I know this is an old bug report, but do you happen to have an original, unmodified copy of the images as they come out of the sensor? Ideally we don't want to have users preprocess the images and here it looks like the dataset you posted has been manually edited, so I can't use it as a reference for troubleshooting. 

I *think* the changes in #1512 should fix this issue, but I can't confirm. Try to reprocess the dataset once #1512 is merged and see if it works with the unmodified images?

sbonaime, 2022-07-28:
@amlobat2 Can you try processing your data with this workflow ?

https://docs.opendronemap.org/multispectral/?highlight=multispectral#sentera-agx710

pierotofy, 2022-11-07:
Closing assuming this should now process; if not, feel free to re-open! :pray:

closed 2022-11-07 by pierotofy