OpenDroneMap/ODM #1295: Strange way of using estimated ground sampling distance

magnus-eigenvision, 2021-06-08:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions and discussion please open a topic on http://community.opendronemap.org/c/opendronemap. 

Please use the format below to report bugs and faults. It will help improve the resolution process. 
****************************************

### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Irrelevant

### What is the problem?

When I look at the code for capping resolution of orthophoto and DEM, I see an unexpected (erroneous?) parameter value.

Here https://github.com/OpenDroneMap/ODM/blob/master/stages/odm_orthophoto.py#L30 and here https://github.com/OpenDroneMap/ODM/blob/master/stages/odm_dem.py#L32 `gsd_error_estimate` gets the unexpected value `-3`. Looking at its usage in `cap_resolution()` here https://github.com/OpenDroneMap/ODM/blob/master/opendm/gsd.py#L79:
```
gsd = gsd * (1 - gsd_error_estimate)
```
this results in multiplying the GSD estimate by 4. Thus we are capping the resolution at one fourth of what seems reasonable based on the GSD estimate.
Why is that?

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

I would expect `gsd_error_estimate` to be something like the default value of 0.1.
If the value `-3` actually is correct, I think there should be some comment in the code explaining it, since it is unintuitive.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Look at the code =)

pierotofy, 2021-06-09:
That's actually a really good question; I don't remember exactly why that `times 4` logic was introduced. I remember we had found that the output size for DEMs was really large for certain datasets (do you remember @smathermather ?), so being GSD an estimate  we opted to reduce the the guess by 4.

Relevant commit that added it:

https://github.com/OpenDroneMap/ODM/commit/57f06c2f6295ccb43922ddac06c8395eb826f3c7

I'm OK fixing this, as it seems an incorrect calculation and I can't find a good reason to keep it as-is. Interested in contributing a pull request @magnus-eigenvision ?

smathermather, 2021-06-10:
The initial decision to investigate ground sampling distance estimation was driven by resource use. The _back of the envelope_  value of having the elevation models run at 1/4 the linear resolution (1/16 effective resolution) was less a factor of resource use and more a factor of ensuring that the elevation model GSD made sense. It is unlikely that for an orthophoto GSD of 1 cm that you'll get good 1 cm elevation model data from the multistereo view step. I think I took a quick look at one of the more mature proprietary tools to see what their default differential was at the time, and it matched what the literature said re: maximum resolution available with minimum noise.

Now all that said, I think this could be made to be 1/2 the linear resolution without issue. The multistereo view approach we use now is fantastic and can probably do a good enough job at this GSD.

smathermather, 2021-06-10:
I think this makes it 1/2 the resolution of the orthophoto, but IDK... : https://github.com/OpenDroneMap/ODM/pull/1299

pierotofy, 2021-06-10:
Ah, makes sense :+1: thanks for chiming in.

I suppose this falls onto user's responsability to choose a good resolution for the DEM? Perhaps then the appropriate change would be to recommend a higher default for `dem-resolution`.

magnus-eigenvision, 2021-06-10:
I see! It makes sense to have lower resolution on DEMs.

But why does the factor between DEM and orthophoto resolution depend on whether the reconstruction is georeferenced? https://github.com/OpenDroneMap/ODM/blob/master/stages/odm_orthophoto.py#L28 Or am I missing something on how the non-georeferenced case works?

#1299 updated odm_dem.py. Should we make sure the value in odm_orthophoto matches? https://github.com/OpenDroneMap/ODM/blob/master/stages/odm_orthophoto.py#L30

The default ratio between DEM and orthophoto resolution is actually not 1/2, but 0.9/2. Could it be a practical problem for some use cases that it's not a nice fraction? One way to handle this would be to use `gsd_error_estimate=0.8`. An even better solution would be to add an argument `gsd_scaling_factor` to `cap_resolution()`, as I think passing a negative argument to `gsd_error_estimate` doesn't make the intent clear, and I think it's kind of hacky ;)

pierotofy, 2021-06-10:
> Should we make sure the value in odm_orthophoto matches?

Yes.

> An even better solution would be to add an argument gsd_scaling_factor to cap_resolution(), as I think passing a negative argument to gsd_error_estimate doesn't make the intent clear, and I think it's kind of hacky ;)

Agree! It's a bit hacky indeed. Feel free to open pull request to improve these?

magnus-eigenvision, 2021-06-10:
I can make a PR.

Can you confirm that the resolution ratio between orthophoto and DEM should be different for georeferenced and non-georeferenced cases? And in that case maybe clarify why? https://github.com/OpenDroneMap/ODM/blob/master/stages/odm_orthophoto.py#L28

pierotofy, 2021-06-10:
We could actually keep the same ratio in the georeferenced vs non-georeferenced case (so fix the existing logic). I chose quite arbitrarily to match the DEMs resolution for non-georeferenced cases.

magnus-eigenvision, 2021-08-23:
Finally! https://github.com/OpenDroneMap/ODM/pull/1333

pierotofy, 2021-08-23:
Closing, thanks @magnus-eigenvision !

closed 2021-08-23 by pierotofy