OpenDroneMap/ODM #1298: GDAL / Error - Can’t load requested DLL

unicorneu, 2021-06-09:
ODM Version: Native Windows Install 1.9.1 Build 18/19

Browser: Firefox

### What is the problem?

Receiving error stating that gdal related dll can't be found. See attached pdf. The directory and files shown in the log does exist. I have GDAL installed and use it with QGIS, which may be triggering the error. As long as I skip the 3d model the task will finish successfully, despite showing the error. If I don't skip the 3D model, however, the task fails.

[webODM_console_GDAL.pdf](https://github.com/OpenDroneMap/ODM/files/6627470/webODM_console_GDAL.pdf)

Saijin-Naib, 2021-06-14:
I believe this was resolved by the user removing their system-wide install of GDAL (which unsets the environment variables that were causing trouble):  
https://community.opendronemap.org/t/gdal-error-cant-load-requested-dll/7871/14?u=saijin_naib

pierotofy, 2022-07-21:
This should be fixed with the latest update. Feel free to re-open if the issue persists. :pray:

closed 2022-07-21 by pierotofy