OpenDroneMap/ODM #1302: Feature Request: Additional Control Point Types

Saijin-Naib, 2021-06-11:
I think it would be very helpful to have a concept within ODM of multiple types of control points.

Currently, we have the concept of 3D Ground Control Points (XYZ). I think a more expanded list would look like this:  

1) Ground Control Point (XY)
1A) For horizontal scaling/registration without high confidence in Z
2) Scale Constraint (XY/XZ/YZ)
2A) To scale the point cloud in a scene given pairs of points with a known distance
3) Manual Tie Point (XY/XYZ)
3A) Manually identified in-situ features
4) Check Point (XY/XYZ)
4A) Used only for Report/verification, not for adjusting reconstruction

Saijin-Naib, 2022-07-11:
I'm not sure where along the line the NaN failure came about, but it is a thing we've seen a few times recently(ish):  
https://community.opendronemap.org/search?expanded=true&q=GCP%20NaN%20order%3Alatest

kikislater, 2023-05-11:
For 2... Useful with dome/banana effect. 
May be a workaround is possible:
- Running complete process
- Measure distance in potree
- Divide measured distance in potree by real measurement
- Add option in ODM to scale model
- Re-run process from beginning using new scale option and use scale value in pdal command. Scale is available in some pdal's writers

Reference: https://github.com/mapillary/OpenSfM/issues/737

DaveWW00, 2025-04-14:
Any updates on this?  At the very least, checkpoints would be nice to add.  The output report giving accuracies of 0.000 for the GCP points is deceiving since its not actually comparing them against anything.  If you had checkpoints you could compare those against the developed model and determine the actual accuracy. Thanks!

smathermather, 2025-07-08:
Related conversation here:
https://community.opendronemap.org/t/scale-constraints-support/24839