OpenDroneMap/ODM #1309: ODM crashes with error from OpenCV: could not find encoder for the specified extension in function 'imencode'

pqvst, 2021-06-22:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system?

Host OS: Ubuntu 18.04.5 LTS
Docker version 19.03.13, build 4484c46d9d
ODM Version: 2.5.7

### What is the problem?

I just updated from ODM 1.0.2 to the latest version 2.5.7. ODM crashes with the following error after about 20 minutes with a dataset of roughly 250 images. I've tested 3 times, same result every time (I even made sure to completely delete the odm output folder before trying again).

> Unspecified error) could not find encoder for the specified extension in function 'imencode'

Partial log output:

```
[INFO]    Undistorting /datasets/project/opensfm ...
2021-06-22 10:09:53,095 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00214.jpg
2021-06-22 10:09:53,095 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00198.jpg
2021-06-22 10:09:53,095 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00222.jpg
2021-06-22 10:09:53,096 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00194.jpg
2021-06-22 10:09:53,096 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00136.jpg
2021-06-22 10:09:53,097 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00099.jpg
2021-06-22 10:09:53,096 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00163.jpg
2021-06-22 10:09:53,096 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00101.jpg
2021-06-22 10:09:53,099 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00184.jpg
2021-06-22 10:09:53,102 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00151.jpg
2021-06-22 10:09:53,104 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00228.jpg
2021-06-22 10:09:53,104 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00216.jpg
2021-06-22 10:09:53,106 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00051.jpg
2021-06-22 10:09:53,097 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00105.jpg
2021-06-22 10:09:53,097 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00211.jpg
2021-06-22 10:09:53,097 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00033.jpg
2021-06-22 10:09:55,686 DEBUG: Undistorting image 2019-10-07_19-30-22 16e-ne_RGB_IMG_00011.jpg
Traceback (most recent call last):
  File "/code/run.py", line 54, in <module>
    retcode = app.execute()
  File "/code/stages/odm_app.py", line 125, in execute
    raise e
  File "/code/stages/odm_app.py", line 89, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 340, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 340, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 340, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 321, in run
    self.process(self.args, outputs)
  File "/code/stages/run_opensfm.py", line 183, in process
    octx.convert_and_undistort(self.rerun(), undistort_callback, image_list_override)
  File "/code/opendm/osfm.py", line 338, in convert_and_undistort
    undistort.run_dataset(ds, "reconstruction.json",
  File "/code/SuperBuild/install/bin/opensfm/opensfm/actions/undistort.py", line 27, in run_dataset
    undistort.undistort_reconstruction_and_images(tracks_manager, r, data, udata, imageFilter)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/undistort.py", line 86, in undistort_reconstruction_and_images
    parallel_map(undistort_image_and_masks, arguments, processes)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/context.py", line 52, in parallel_map
    res = Parallel(batch_size=batch_size)(delayed(func)(arg) for arg in args)
  File "/usr/local/lib/python3.8/dist-packages/joblib/parallel.py", line 1061, in __call__
    self.retrieve()
  File "/usr/local/lib/python3.8/dist-packages/joblib/parallel.py", line 940, in retrieve
    self._output.extend(job.get(timeout=self.timeout))
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 771, in get
    raise self._value
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/usr/local/lib/python3.8/dist-packages/joblib/_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/joblib/parallel.py", line 262, in __call__
    return [func(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/joblib/parallel.py", line 262, in <listcomp>
    return [func(*args, **kwargs)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/undistort.py", line 104, in undistort_image_and_masks
    udata.save_undistorted_image(k, v)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/dataset.py", line 1105, in save_undistorted_image
    self.io_handler.imwrite(self._undistorted_image_file(image), array)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/io.py", line 1371, in imwrite
    imwrite_from_fileobject(fwb, image, path)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/io.py", line 1191, in imwrite_from_fileobject
    _, im_buffer = cv2.imencode(ext, image)
cv2.error: OpenCV(4.5.0) /code/SuperBuild/src/opencv/modules/imgcodecs/src/loadsave.cpp:941: error: (-2:Unspecified error) could not find encoder for the specified extension in function 'imencode'
```

Running ODM in docker. Using the following args:

```
--ignore-gsd --dsm --cog --project-path /datasets project
```

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

ODM shouldn't crash. Exact same flight worked fine in ODM 1.0.2

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I can provide a copy of the dataset if needed.

pierotofy, 2021-06-22:
Thanks for the report. Seems to be caused by the space in the filenames. As a workaround, don't use spaces until a fix is released (or open a pull request?). :pray:

pierotofy, 2021-06-22:
Should be fixed with https://github.com/OpenDroneMap/OpenSfM/commit/578b0636b1955ed971b406a397c7d8e709e04877

closed 2021-06-22 by pierotofy