OpenDroneMap/ODM #1312: PoissonRecon failed with 1 threads

sbonaime, 2021-06-29:
Odm fails at PoissonRecon step.

my command is
````
docker run -it --rm -v  "/data/my_images/:/code/images" -v "$(pwd)/odm_georeferencing:/code/odm_georeferencing"     -v "$(pwd)/odm_meshing:/code/odm_meshing"     -v "$(pwd)/odm_dem:/code/odm_dem"     -v "$(pwd)/odm_orthophoto:/code/odm_orthophoto"     -v "$(pwd)/odm_texturing:/code/odm_texturing"     -v "$(pwd)/odm_report:/code/odm_report" opendronemap/odm --orthophoto-png --mesh-octree-depth 16 --ignore-gsd --dtm   --smrf-threshold 0.4 --smrf-window 24 --dsm --pc-csv --pc-las --orthophoto-kmz   --ignore-gsd --feature-quality high --max-concurrency 16  --use-hybrid-bundle-adjustment --build-overviews --time --min-num-features 10000
````

Linux  4.19.0-16-amd64 #1 SMP Debian 4.19.181-1 (2021-03-19) x86_64 GNU/Linux

- Docker opendronemap/odm   latest    409551646e8d
  20_04_21_X7_Chambon_All docker image inspect 409551646e8d
        "Id": "sha256:409551646e8d301bcfc1de8bbd0dfba5516ba944366a033dfee18953f337f95c",
        "RepoTags": [
            "opendronemap/odm:latest"
        ],
        "RepoDigests": [
            "opendronemap/odm@sha256:7e54c74fb6b040da60effbdf98980f875337e4b5c8027bc939e3d446703795a7"
        ],
        "Created": "2021-06-22T17:31:39.210863835Z",

````
[INFO]    Merging 52 point cloud chunks to /code/odm_filterpoints/point_cloud.ply
[INFO]    Finished odm_filterpoints stage
[INFO]    Running odm_meshing stage
[INFO]    Writing ODM Mesh file in: /code/odm_meshing/odm_mesh.ply
[INFO]    running "/code/SuperBuild/install/bin/PoissonRecon" --in "/code/odm_filterpoints/point_cloud.ply" --out "/code/odm_meshing/odm_mesh.dirty.ply" --depth 16 --pointWeight 4.0 --samplesPerNode 1.0 --threads 15 -
-maxMemory 94 --bType 2 --linearFit
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
[WARNING] Child returned 134
WARNING] PoissonRecon failed with 7 threads, let's retry with 3...
[INFO]    running "/code/SuperBuild/install/bin/PoissonRecon" --in "/code/odm_filterpoints/point_cloud.ply" --out "/code/odm_meshing/odm_mesh.dirty.ply" --depth 16 --pointWeight 4.0 --samplesPerNode 1.0 --threads 7 --
maxMemory 93 --bType 2 --linearFit
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
----Aborted (core dumped)
[WARNING] Child returned 134
[WARNING] PoissonRecon failed with 3 threads, let's retry with 1...
[INFO]    running "/code/SuperBuild/install/bin/PoissonRecon" --in "/code/odm_filterpoints/point_cloud.ply" --out "/code/odm_meshing/odm_mesh.dirty.ply" --depth 16 --pointWeight 4.0 --samplesPerNode 1.0 --threads 3 --
maxMemory 94 --bType 2 --linearFit
---terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
[WARNING] Child returned 134
[WARNING] PoissonRecon failed with 1 threads, let's retry with 0...
[INFO]    running "/code/SuperBuild/install/bin/PoissonRecon" --in "/code/odm_filterpoints/point_cloud.ply" --out "/code/odm_meshing/odm_mesh.dirty.ply" --depth 16 --pointWeight 4.0 --samplesPerNode 1.0 --threads 1 --
maxMemory 94 --bType 2 --linearFit
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
[WARNING] Child returned 134
[INFO]    running "/code/SuperBuild/install/bin/OpenMVS/ReconstructMesh" -i "/code/odm_meshing/odm_mesh.dirty.ply" -o "/code/odm_meshing/odm_mesh.ply" --remove-spikes 0 --remove-spurious 20 --smooth 0 --target-face-nu
m 400000
10:34:33 [App     ] Build date: Jun 22 2021, 17:22:42
10:34:33 [App     ] CPU: Intel(R) Xeon(R) CPU E5-2650 0 @ 2.00GHz (16 cores)
10:34:33 [App     ] RAM: 125.86GB Physical Memory 975.00MB Virtual Memory
10:34:33 [App     ] OS: Linux 4.19.0-16-amd64 (x86_64)
10:34:33 [App     ] SSE & AVX compatible CPU & OS detected
10:34:33 [App     ] Command line: -i /code/odm_meshing/odm_mesh.dirty.ply -o /code/odm_meshing/odm_mesh.ply --remove-spikes 0 --remove-spurious 20 --smooth 0 --target-face-num 400000

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
  File "/code/stages/odm_app.py", line 89, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 340, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 340, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 340, in run
    self.next_stage.run(outputs)
  [Previous line repeated 3 more times]
  File "/code/opendm/types.py", line 321, in run
    self.process(self.args, outputs)
  File "/code/stages/odm_meshing.py", line 24, in process
    mesh.screened_poisson_reconstruction(tree.filtered_point_cloud,
 File "/code/opendm/mesh.py", line 207, in screened_poisson_reconstruction
    system.run('"{reconstructmesh}" -i "{infile}" '
  File "/code/opendm/system.py", line 106, in run
    raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying.html
`````

sbonaime, 2021-06-29:
Maybe it is linked with 
https://github.com/mkazhdan/PoissonRecon/issues/184
and
https://github.com/mkazhdan/PoissonRecon/issues/105
I will try with 
`--mesh-octree-depth 12`
instead of
`--mesh-octree-depth 16`

sbonaime, 2021-06-29:
I don't have the issue with 
```
--mesh-octree-depth 12
```

smathermather, 2021-06-30:
I missed this. Yes, mesh depth should probably almost never exceed 12, maybe 13 max. 16 would certainly crash.

Perhaps we should catch extra high mesh depth values and warn while enforcing a ceiling. Can you change the issue to a feature request to that effect?

sbonaime, 2021-07-02:
@smathermather thanks. I don't know how to change this issue to  a "feature request".

smathermather, 2021-07-02:
Just change the title to "Feature request: " amd them retitle the rest to briefly describe what it should do. In this case, something like "Feature request: limit max octree depth on mesh generation" would be ok.

PR #1313 (2021-07-03): Fix COG compression, cap mesh-octree-depth values
Fixes the problem reported on https://community.opendronemap.org/t/can-not-disable-cog-option-cloud-optimized-geotiffs-in-webodm/8184

Closes #1312﻿
closed 2021-07-03 by pierotofy