OpenDroneMap/ODM #1315: Build script (configure.sh install) fails at PoissonRecon IF RAM is below 16GB

tteke, 2021-07-08:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions and discussion please open a topic on http://community.opendronemap.org/c/opendronemap. 

Please use the format below to report bugs and faults. It will help improve the resolution process. 
****************************************

### How did you install OpenDroneMap? (Docker, natively, ...)?

natively

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

ubuntu 20.04

### What is the problem?

build script fails during the PoissonRecon build step. If the computer has 8GB ram. Failed it on 2 different computers, then upgraded ram to 16GB on one of them. 16GB Ram system built successfully.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

NA

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Try building ODM (2.5.7, 2.5.6, 2.5.5 only tried those 3) on a ubuntu 20.04 system with 8GB ram

pierotofy, 2021-07-08:
Noticed that too, we went around it by allocating extra swap. If you have any suggestions on how to improve it, we'd welcome a pull request. This started happening with the newer versions of the PoissonRecon code. Older versions compiled fine with less memory.

tteke, 2021-07-08:
Got it. Maybe I should open an issue on their repo then. Do you know by chance exactly which release started causing the issue ?

pierotofy, 2021-07-09:
Not sure which one; but within the last 2 years if that helps narrow it down :grimacing:

closed 2021-10-11 by pierotofy

sbonaime, 2023-07-27:
same problem as #1679 fix with #1681