OpenDroneMap/ODM #1317: ODM Console not working when installed in Program Files

casbolwerk, 2021-07-28:
### How did you install OpenDroneMap? (Docker, natively, ...)?

The native releases download on windows

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 90 on windows 10

### What is the problem?

The pyvenv.cfg file (and possibly others) are not created when ODM is installed in a path with a space in it. The issue lies with the win32env.bat file, which is supposed to create this pyvenv.cfg file. It does not handle paths with spaces and ends up not creating it

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

win32env.bat should be able to handle spaces in paths. It goes wrong at the two lines 24 and 25 in win32env.bat.
```
echo home = %ODMBASE%\python38> %PYENVCFG%
echo include-system-site-packages = false>> %PYENVCFG%
```
I tried putting quotation marks around the paths like `"%PYENVCFG%"`, but while creating the pyvenv.cfg file it still got errors because the python code could not handle the spaces in `%ODMBASE%\python38`. This might be a bigger issue than just creating the pyvenv.cfg file.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Install ODM inside a path with a space (Program Files is a popular one on windows) and try to run it. It will say `No pyvenv.cfg file` in the console. Then go ahead and change win32env.bat.

pierotofy, 2021-12-23:
This is fixed in `2.7.0`. :clinking_glasses:

closed 2021-12-23 by pierotofy

coumbsek, 2022-03-18:
After updating from 2.6.7 to 2.8.0 I still observe this. I have check and did not commit the correct changes last time.
We must add quotes around %PYENVCFG% line 24 and 25 during the hot-patching of pyvenv.cfg.
`echo home = %ODMBASE%\python38> "%PYENVCFG%"`
`echo include-system-site-packages = false>> "%PYENVCFG%"`

How to you want to handle this ? Should I make a pull request just for this ?

pierotofy, 2022-03-18:
A PR would be most welcome!