OpenDroneMap/ODM #1332: Using SIFT without GPS information

nikhil-aren, 2021-08-20:
Hi, I noticed that if we do not have GPS information, the code currently forces us to use BOW matching with HAHOG ("osfm.py", lines 205-207). However, I would like to use SIFT with brute force matching. From looking at the github history, this was put in place before SIFT was supported. Is there a reason for using BOW matching + HAHOG when there is not GPS information?

```
if not has_gps:
    log.ODM_INFO("No GPS information, using BOW matching")
    use_bow = True
```

pierotofy, 2021-08-21:
The reason this code block was introduced is that the absence of GPS information forces OpenSfM to run the matcher on all image pairs, which is slow(er) with FLANN, but faster with BoW (or at least it used to be, FLANN has become faster as of late).

I would have no problem merging a PR that removes this logic, since one can control the matcher type via `--matcher-type`.

pierotofy, 2021-11-15:
This is fixed in #1362. If you explictly pass `--matcher-type` you can now bypass the `bow` default. :pray:

closed 2021-11-15 by pierotofy