OpenDroneMap/ODM #1339: Hello, I got a a strange error code. And I just can not access to OpenDroneMap Community, how to fix it?

JimXu1989, 2021-09-06:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions and discussion please open a topic on http://community.opendronemap.org/c/opendronemap. 

Please use the format below to report bugs and faults. It will help improve the resolution process. 
****************************************

### How did you install OpenDroneMap? (Docker, natively, ...)?

[docker]

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[chrome]

### What is the problem?

[ Every going nice, and most of the work is done, while an error occure:
[INFO]    Writing MVS Textured file in: /datasets/project/odm_texturing/odm_textured_model_geo.obj
[INFO]    running "/code/SuperBuild/install/bin/texrecon" "/datasets/project/opensfm/undistorted/reconstruction.nvm" "/datasets/project/odm_meshing/odm_mesh.ply" "/datasets/project/odm_texturing/odm_textured_model_geo" -d gmi -o gauss_clamping -t none --no_intermediate_results      
/code/SuperBuild/install/bin/texrecon: error while loading shared libraries: libtbb.so.2: cannot open shared object file: No such file or directory

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 127
Traceback (most recent call last):
  File "/code/stages/odm_app.py", line 94, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 340, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 340, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 340, in run
    self.next_stage.run(outputs)
  [Previous line repeated 4 more times]
  File "/code/opendm/types.py", line 321, in run
    self.process(self.args, outputs)
  File "/code/stages/mvstex.py", line 108, in process
    system.run('"{bin}" "{nvm_file}" "{model}" "{out_dir}" '
  File "/code/opendm/system.py", line 106, in run
    raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 127

===== Done, human-readable information to follow... =====

[ERROR]   The program exited with a strange error code. Please report it at https://community.opendronemap.org

It seem like some thing wrong with tbb, so I sudo apt-get install libtbb2, and I can find libtbb.so.2 in my computer, but the error still occure.

And when  I want to access the ODM community,  I found I just cannot access to it.
]

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

[Type answer here]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[Type answer here]

APOS80, 2021-09-06:
I have the same problem

pierotofy, 2021-09-06:
Hey @JimXu1989 :hand: thanks for the report! Are you running the latest version? Could you try to run `docker pull opendronemap/odm` ?

pierotofy, 2021-09-06:
I'm assuming this is the opendronemap/odm:gpu instance.

If so, it should be fixed with https://github.com/OpenDroneMap/ODM/commit/1915afe38001f938b3b5a205d4b6e310c711b7e7

Wait a few hours, update then retry? :pray:

JimXu1989, 2021-09-07:
> I'm assuming this is the opendronemap/odm:gpu instance.
> 
> If so, it should be fixed with [1915afe](https://github.com/OpenDroneMap/ODM/commit/1915afe38001f938b3b5a205d4b6e310c711b7e7)
> 
> Wait a few hours, update then retry?
Thanks for your reply!!
Thanks for the great work!
Yes, I use the gpu instance, I will try later

JimXu1989, 2021-09-07:
> I'm assuming this is the opendronemap/odm:gpu instance.
> 
> If so, it should be fixed with [1915afe](https://github.com/OpenDroneMap/ODM/commit/1915afe38001f938b3b5a205d4b6e310c711b7e7)
> 
> Wait a few hours, update then retry?

Hi, I retried again, and I got another error:

docker run -ti --rm -v c:/Users/youruser/datasets:/datasets --gpus all opendronemap/odm:gpu --project-path /datasets/project
docker: Error response from daemon: invalid mode: /datasets.
See 'docker run --help'.

JimXu1989, 2021-09-07:
> I'm assuming this is the opendronemap/odm:gpu instance.
> 
> If so, it should be fixed with [1915afe](https://github.com/OpenDroneMap/ODM/commit/1915afe38001f938b3b5a205d4b6e310c711b7e7)
> 
> Wait a few hours, update then retry?

And how to build natively with GPU?
Thanks!

closed 2021-09-09 by pierotofy