OpenDroneMap/ODM #1340: Strange output

APOS80, 2021-09-06:
### How did you install OpenDroneMap? (Docker, natively, ...)?

WebODM windows and nodeODM in docker in ubuntu

### What's your browser and operating system? 

Chrome windows/Linux

### What is the problem?

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

[Type answer here]

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[Type answer here]

APOS80, 2021-09-06:
Projections are flat so if ODM gets GCP’s in a flat system I could just treat it like a local system that’s flat and put the EPSG tag from the EPSG list on the output.

pierotofy, 2022-03-08:
The output should of ODM always be in the nearest UTM CRS (even if your GCP uses a different projection/CRS).

If you think the output is incorrect, please provide at a minimum a dataset (with the GCP), screenshots and an explanation of which parts of the output you think are incorrect (show measurements and how you calculated them with expected vs actual)?  :pray: 

I can re-open this once we get more information and can confirm that this is a bug.