OpenDroneMap/ODM #1341: Feature Request: Option to determine --gps-accuracy and then pass that to pipeline for processing

Saijin-Naib, 2021-09-07:
### How did you install OpenDroneMap? (Docker, natively, ...)?
NA

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
NA

### What is the problem?
Having an accurate assessment of the --gps-accuracy for a particular collect is pretty difficult, especially on non-RKT/PPK platforms.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
I think, given our improved reporting, that we should have an option for ODM to self-optimize the --gps-accuracy flag early in the pipeline when it starts doing the reconstruction and determines that value. Then, it should pass that value back into the pipeline and reprocess what it needs to with that new --gps-accuracy value forward.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Use the default 10m on a dataset with better --gps-accuracy.   
Read the report, note the GPS Error value.  
Re-process that same dataset using the GPS Error from the report (or I round up to the next whole meter to not over-constrain).  
 Read the report.  
Observe that the GPS Error has shrunk even further.

I think this could be critical for helping us reduce our real/reported error, especially in situations where folks are A/B testing our pipeline against another pipeline/report and don't understand why our processing seems to indicate higher error/imprecision.

pierotofy, 2021-09-07:
You have to be a bit careful here; without ground control points it's difficult to compute a good absolute error, and the value outputted by the report in absence of GCPs assumes that your GPS measurements are precise (0 uncertainty), so almost as good as ground control points (which typically aren't) for the purposes of calculating an absolute error. But any uncertainty in your GPS measurements should be added to the value output by the report.

I debated adding the uncertainty directly into the report value, except we *typically* (read: most consumer drones) don't really tell us the true accuracy of GPS shots, so we would either overestimate or underestimate the actual uncertainty by naively picking an estimate. I haven't found good literature that outlines possible approaches to incorporate GPS uncertainty into the absolute error estiamte, although I have ideas.

Point is, constraining `--gps-accuracy` might change the output value in the report, but doesn't really make a dataset more accurate.

Saijin-Naib, 2021-09-07:
Makes total sense.

Ah, no free lunch.

pierotofy, 2021-09-07:
I should have added, this doesn't invalidate the idea for a dynamic gps-accuracy, it might be the case that such a mechanism would improve reconstructions overall, I just don't think it will help improve accuracy. :) So maybe it's a free snack.

Saijin-Naib, 2021-09-07:
What I have not checked yet, but I need to, is if it helps constrain the center-point to something more accurate to ground truth.

We've had folks on the forum discussing pretty big offsets from our reconstruction vs other products and my naive first-blush on this is that this might mitigate that some, whether or not it is just over-fitting.

I have to dig through my Solo data... I think the DOP and GPS Precision get burned into each image. Perhaps like you said, if these are present they may help.

YanNoun, 2021-11-03:
@pierotofy OpenSfM can now compute the bias of the GPS sensor when you have GCPs, with the option `bundle_compensate_gps_bias` in `config.yaml`. Computed bias is also outputted in the report.

pierotofy, 2022-07-26:
Closing since we can use `bundle_compensate_gps_bias` :+1:

closed 2022-07-26 by pierotofy