OpenDroneMap/ODM #1348: Error: Ply file does not contain normals

DonkeyLiao, 2021-09-14:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What is the problem?

I use 158 pictures to merge TIFF and it run success, **the command is:** docker run -i --rm -v /www/wwwroot/datasets:/datasets opendronemap/odm --project-path /datasets 3dmesh --fast-orthophoto --pc-quality lowest --feature-quality lowest --skip-report. I also got the odm_textured_model_geo.obj file under the odm_texturing_25d folder.

After that I use the same 158 pictures to build 3D Textured Models but I got a Error: Ply file does not contain normals, **I use the command:** docker run -i --rm -v /www/wwwroot/datasets:/datasets opendronemap/odm --project-path /datasets 3dmesh --fast-orthophoto --pc-quality lowest --feature-quality lowest --skip-report --use-3dmesh. How can I fix it?

DonkeyLiao, 2021-09-24:
I use the option 'use-3dmesh' only and then run success, it seems that there some options conflict.

DonkeyLiao, 2021-09-24:
emmm, the 3D Textured Models is a .obj file, can  ODM generate a 3d tiles file or .osgb file?

pierotofy, 2021-09-24:
Hey @DonkeyLiao :hand: this should be fixed with https://github.com/OpenDroneMap/ODM/commit/97cd395a2ace7f31392b4b9a1124a1acd02502cf try to update?

For all other questions, let's move the conversation over to https://community.opendronemap.org :pray: 

If the issue is not resolved, feel free to re-open this issue. Thanks!

closed 2021-09-24 by pierotofy