OpenDroneMap/ODM #1349: Skip corrupted images

pierotofy, 2021-09-16:
ODM should probably ignore (emit a warning) corrupted images and process whatever it can instead of stopping.

Saijin-Naib, 2021-09-16:
Helps/works with Issue #1326

pierotofy, 2021-11-17:
A common error that happens with corrupted images:

```
[INFO]    texturing_skip_local_seam_leveling: False
[INFO]    texturing_tone_mapping: none
[INFO]    tiles: False
[INFO]    time: False
[INFO]    use_3dmesh: False
[INFO]    use_exif: False
[INFO]    use_fixed_camera_params: False
[INFO]    use_hybrid_bundle_adjustment: False
[INFO]    verbose: False
[INFO]    ==============
[INFO]    Running dataset stage
[INFO]    Loading dataset from: /var/www/data/2367b185-148f-45bb-a146-84ad688c0adb/images
[INFO]    Loading 192 images
[WARNING] Cannot read /var/www/data/2367b185-148f-45bb-a146-84ad688c0adb/images/MAX_0590.JPG with PIL, fallback to cv2: cannot identify image file '/var/www/data/2367b185-148f-45bb-a146-84ad688c0adb/images/MAX_0590.JPG'
Traceback (most recent call last):
File "/code/opendm/get_image_size.py", line 13, in get_image_size
with Image.open(file_path) as img:
File "/usr/local/lib/python3.9/dist-packages/PIL/Image.py", line 2943, in open
raise UnidentifiedImageError(
PIL.UnidentifiedImageError: cannot identify image file '/var/www/data/2367b185-148f-45bb-a146-84ad688c0adb/images/MAX_0590.JPG'

During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/code/run.py", line 53, in <module>
retcode = app.execute()
File "/code/stages/odm_app.py", line 130, in execute
raise e
File "/code/stages/odm_app.py", line 94, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 328, in run
self.process(self.args, outputs)
File "/code/stages/dataset.py", line 112, in process
p = types.ODM_Photo(f)
File "/code/opendm/photo.py", line 92, in __init__
self.parse_exif_values(path_file)
File "/code/opendm/photo.py", line 274, in parse_exif_values
self.width, self.height = get_image_size.get_image_size(_path_file)
File "/code/opendm/get_image_size.py", line 19, in get_image_size
width = img.shape[1]
AttributeError: 'NoneType' object has no attribute 'shape'
```

closed 2021-12-18 by pierotofy