OpenDroneMap/ODM #1352: Why multiply -1 to self.altitude??

gomajirosan, 2021-10-05:
https://github.com/OpenDroneMap/ODM/blob/5cb332ae7b163236919f1af737c8b4f1f8c6e5f6/opendm/photo.py#L138

According to 
https://www.awaresystems.be/imaging/tiff/tifftags/privateifd/gps/gpsaltituderef.html

GPSAltitudeRef
0 = Above sea level
1 = Below sea level

and a value of 1 is given and the altitude is indicated as an absolute value in the GPSAltitude tag

Why multiplying -1 to self.altitude ??

pierotofy, 2021-10-05:
GPSAltitude is a positive integer.

If the value is "below sea level" you need to make that value negative.

pierotofy, 2021-10-06:
> The exif is not correct?

Possibly! 

What does `exiftool` say? https://exiftool.org/

gomajirosan, 2021-10-06:
{'xmp:ModifyDate': '2018-12-11', 'xmp:CreateDate': '2018-12-11', 'xmp:CreatorTool': 'v01.09.2222', 'dc:format': 'image/jpg', 'dc:description': '', 'dc:description[1]': 'DCIM\\\\100MEDIA\\\\DJI_0506.JPG', 'dc:description[1]/?xml:lang': 'x-default', 'drone-dji:AbsoluteAltitude': '-89.27', 'drone-dji:RelativeAltitude': '+25.00', 'drone-dji:GimbalRollDegree': '+0.00', 'drone-dji:GimbalYawDegree': '+4.70', 'drone-dji:GimbalPitchDegree': '-89.10', 'drone-dji:FlightRollDegree': '+3.20', 'drone-dji:FlightYawDegree': '+5.90', 'drone-dji:FlightPitchDegree': '-0.80', 'drone-dji:CamReverse': '0', 'drone-dji:GimbalReverse': '0', 'drone-dji:SelfData': 'Undefined', 'drone-dji:CalibratedFocalLength': '4545.454590', 'drone-dji:CalibratedOpticalCenterX': '2640.000000', 'drone-dji:CalibratedOpticalCenterY': '1978.000000', 'drone-dji:RtkFlag': '0', 'crs:Version': '7.0', 'crs:HasSettings': 'False', 'crs:HasCrop': 'False', 'crs:AlreadyApplied': 'False', 'tiff:Orientation': '1', 'tiff:XResolution': '72/1', 'tiff:YResolution': '72/1', 'tiff:ResolutionUnit': '2', 'tiff:YCbCrPositioning': '1', 'tiff:Make': 'DJI', 'tiff:Model': 'FC6520', 'exif:ColorSpace': '1', 'exif:CompressedBitsPerPixel': '9311521/2610960', 'exif:PixelXDimension': '5280', 'exif:PixelYDimension': '3956', 'exif:ExposureTime': '1562/1000000', 'exif:FNumber': '170/100', 'exif:ExposureProgram': '4', 'exif:ShutterSpeedValue': '-9312/-1000', 'exif:ApertureValue': '153/100', 'exif:ExposureBiasValue': '0/32', 'exif:MaxApertureValue': '153/100', 'exif:SubjectDistance': '0/100', 'exif:MeteringMode': '3', 'exif:LightSource': '10', 'exif:FocalLength': '1500/100', 'exif:ExposureIndex': '0/0', 'exif:CustomRendered': '0', 'exif:ExposureMode': '0', 'exif:WhiteBalance': '1', 'exif:DigitalZoomRatio': '0/0', 'exif:FocalLengthIn35mmFilm': '30', 'exif:SceneCaptureType': '0', 'exif:GainControl': '0', 'exif:Contrast': '0', 'exif:Saturation': '0', 'exif:Sharpness': '0', 'exif:SubjectDistanceRange': '0', 'exif:GPSAltitudeRef': '1', 'exif:GPSAltitude': '89269/1000', 'exif:DateTimeOriginal': '2018-12-11T14:00:57', 'exif:ISOSpeedRatings': '', 'exif:ISOSpeedRatings[1]': '120', 'exif:ExifVersion': '0230', 'exif:FlashpixVersion': '0010', 'exif:ComponentsConfiguration': '', 'exif:ComponentsConfiguration[1]': '0', 'exif:ComponentsConfiguration[2]': '3', 'exif:ComponentsConfiguration[3]': '2', 'exif:ComponentsConfiguration[4]': '1', 'exif:Flash': '', 'exif:Flash/exif:Fired': 'False', 'exif:Flash/exif:Return': '0', 'exif:Flash/exif:Mode': '0', 'exif:Flash/exif:Function': 'True', 'exif:Flash/exif:RedEyeMode': 'False', 'exif:FileSource': '3', 'exif:SceneType': '1', 'exif:GPSVersionID': '2.3.0.0', 'exif:GPSLatitude': '36,17.0493N', 'exif:GPSLongitude': '140,28.2417E', 'exifEX:BodySerialNumber': 'f90b4515da29a84def1324aa20373743', 'exifEX:LensMake': 'UnKnown', 'exifEX:LensModel': 'DJI MFT 15mm F1.7 ASPH', 'exifEX:PhotographicSensitivity': '120', 'aux:SerialNumber': 'f90b4515da29a84def1324aa20373743', 'aux:Lens': 'DJI MFT 15mm F1.7 ASPH', 'photoshop:DateCreated': '2018-12-11T14:00:57', 'file_path': '***', 'Make': 'DJI', 'datetime': '2018-12-11T14:00:57', 'Model': 'FC6520', 'digital_zoom_ratio': -9999, 'lat_deg': 36.284155, 'lon_deg': 140.470695, 'rel_alt_m': 25.0, 'gimbal_pit_deg': -89.1, 'gimbal_rol_deg': 0.0, 'gimbal_yaw_deg': 4.7, 'flight_pit_deg': -0.8, 'flight_rol_deg': 3.2, 'flight_yaw_deg': 5.9, 'focal_length_35mm': 30.0, 'fnumber': 1.7, 'shutter_speed': -9999, 'focal_length_mm': -9999, 'image_width': 5280, 'image_height': 3956}"

I confirmed that there is no problem in 
https://github.com/OpenDroneMap/ODM/blob/5cb332ae7b163236919f1af737c8b4f1f8c6e5f6/opendm/photo.py#L138

One more curious thing is I compared the result of 2.6.1 and 1.0.2 and the output DSM value range not the same.
It could happens?

[image]