OpenDroneMap/ODM #1366: radiometric-calibration doesn't work properly

lucadelu, 2021-10-30:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker using webodm on linux server

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox/Chromium on Linux

### What is the problem?

Using Multispectral images coming from Mapir Camera and choosing `radiometric-calibration: camera` I'm not able to process the data.

The problem seems during the opensfm stage because it is looking for a .tif file but instead it is a .JPG image

~~~
[INFO]    running /code/SuperBuild/install/bin/opensfm/bin/opensfm export_visualsfm --points "/var/www/data/34a4d273-e578-49ca-a7fe-3feda8bc6217/opensfm"
Traceback (most recent call last):
File "/code/SuperBuild/install/bin/opensfm/opensfm/io.py", line 1221, in image_size_from_fileobject
with Image.open(fb) as img:
File "/usr/local/lib/python3.9/dist-packages/PIL/Image.py", line 2943, in open
raise UnidentifiedImageError(
PIL.UnidentifiedImageError: cannot identify image file <_io.BufferedReader name='/var/www/data/34a4d273-e578-49ca-a7fe-3feda8bc6217/opensfm/undistorted/images/2021_0922_145250_210.JPG.tif'>
~~~

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Finishing the procedure for the reconstruction

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

You can try with this dataset https://drive.google.com/file/d/1sa-P_VzhsuGrAw4gb6n5ePU5qhDqm-zj/view?usp=sharing
I also attach the full log
[console.txt](https://github.com/OpenDroneMap/ODM/files/7445937/console.txt)

Gurkengewuerz, 2021-12-04:
Hey @lucadelu i tried the exact same thing. I opened a community thread last night and hopefully get a response. Did you get any further? Maybe you can also post more details at the community post. 

https://community.opendronemap.org/t/odm-stops-processing-with-cannot-identify-image-file/9739

lucadelu, 2021-12-04:
Hi @Gurkengewuerz no I was not able to go further, I saw your pull request in OpenSfM, it seems that one the problem. Thanks

pierotofy, 2022-01-04:
This should be fixed with the latest changes in OpenSfM / ODM. 🙏 

Thanks!

lucadelu, 2022-01-07:
@pierotofy I tried to update my webodm  instance (with "./webodm.sh update") but I got the same error, should the patch already be applied?

pierotofy, 2022-01-08:
Mm, maybe I closed this one prematurely (I thought it was fixed, but perhaps not, now that I look at the stack trace more carefully).

Gurkengewuerz, 2022-01-08:
Weird. these were the changes which were enough for me. My dataset is still online on the forum for testing if someone is interesting. I recently reinstalled my computer so i still have to setup my enviroment to test the update.

pierotofy, 2022-01-12:
I'm getting a different error:

```
da-device -1
18:19:53 [App     ] error: failed loading image header
18:19:53 [App     ] error: failed loading image header
18:19:53 [App     ] error: failed loading image header
18:19:53 [App     ] error: failed reloading image '/datasets/nir/opensfm/undistorted/images/2021_0922_144540_038.JPG.tif'
18:19:53 [App     ] error: failed reloading image '/datasets/nir/opensfm/undistorted/images/2021_0922_144540_038.JPG.tif'
18:19:53 [App     ] error: failed loading image header
18:19:53 [App     ] error: failed reloading image '/datasets/nir/opensfm/undistorted/images/2021_0922_144543_039.JPG.tif'
18:19:53 [App     ] error: failed reloading image '/datasets/nir/opensfm/undistorted/images/2021_0922_144543_039.JPG.tif'
18:19:53 [App     ] error: failed reloading image '/datasets/nir/opensfm/undistorted/images/2021_0922_144613_051.JPG.tif'
18:19:53 [App     ] error: failed loading image header
18:19:53 [App     ] error: failed loading image header
18:19:53 [App     ] error: failed reloading image '/datasets/nir/opensfm/undistorted/images/2021_0922_144553_043.JPG.tif'
18:19:53 [App     ] error: preparing images for dense reconstruction failed (errors loading images)

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
  File "/code/stages/openmvs.py", line 91, in process
    run_densify()
  File "/code/stages/openmvs.py", line 86, in run_densify
    system.run('"%s" "%s" %s' % (context.omvs_densify_path,
  File "/code/opendm/system.py", line 106, in run
    raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/code/stages/odm_app.py", line 94, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 346, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 346, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 346, in run
    self.next_stage.run(outputs)
  [Previous line repeated 1 more time]
  File "/code/opendm/types.py", line 327, in run
    self.process(self.args, outputs)
  File "/code/stages/openmvs.py", line 98, in process
    run_densify()
  File "/code/stages/openmvs.py", line 86, in run_densify
    system.run('"%s" "%s" %s' % (context.omvs_densify_path,
  File "/code/opendm/system.py", line 106, in run
    raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1
```

@lucadelu can you confirm this is the same error you're getting? The original one:

```
[INFO]    running /code/SuperBuild/install/bin/opensfm/bin/opensfm export_visualsfm --points "/var/www/data/34a4d273-e578-49ca-a7fe-3feda8bc6217/opensfm"
Traceback (most recent call last):
File "/code/SuperBuild/install/bin/opensfm/opensfm/io.py", line 1221, in image_size_from_fileobject
with Image.open(fb) as img:
File "/usr/local/lib/python3.9/dist-packages/PIL/Image.py", line 2943, in open
raise UnidentifiedImageError(
PIL.UnidentifiedImageError: cannot identify image file <_io.BufferedReader name='/var/www/data/34a4d273-e578-49ca-a7fe-3feda8bc6217/opensfm/undistorted/images/2021_0922_145250_210.JPG.tif'>
```

Doesn't seem to be triggered.

pierotofy, 2022-01-12:
The key issue is that these images are RGB; ODM does not know how to apply radiometric calibration on RGB images.

What do the RGB channels represent here?

lucadelu, 2022-01-13:
> I'm getting a different error:
>
> ```
> da-device -1
> 18:19:53 [App     ] error: failed loading image header
> 18:19:53 [App     ] error: failed loading image header
> 18:19:53 [App     ] error: failed [...]

yes, you are right, this is the error,

> ```
> [INFO]    running /code/SuperBuild/install/bin/opensfm/bin/opensfm export_visualsfm --points "/var/www/data/34a4d273-e578-49ca-a7fe-3feda8bc6217/opensfm"
> Traceback (most recent call last):
> File [...]

yes, you are right, the original problem is solved. should I open new ticket for the new problem and close this one?

lucadelu, 2022-01-13:
> The key issue is that these images are RGB; ODM does not know how to apply radiometric calibration on RGB images.
> 
> What do the RGB channels represent here?

They are not RGB but Red - nothing or something else - NIR

pierotofy, 2022-01-13:
The JPGs in the linked dataset are RGB (8bit, 3 channels).

ninsbl, 2022-01-26:
Had a similar (not sure it is the same) issue with another drone. Solved it by copying out single band TIF files for processing (so only TIF files and no JPG in the images drectory). See here for reference:
https://community.opendronemap.org/t/dji-phantom-4-pro-multipectral-w-o-rtk/4978/18

pierotofy, 2022-07-26:
Should be fixed with #1512 

[image]

closed 2022-07-26 by pierotofy