OpenDroneMap/ODM #1368: Possible Bug: --matcher-type does not seem to override opensfm config.py

Saijin-Naib, 2021-11-09:
### How did you install OpenDroneMap? (Docker, natively, ...)?
Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
NA

### What is the problem?
Currently, the config.py for OpenSFM has a number of parameters set that are over-ridden by some parameters passed to ODM/WebODM. In the case of:  
```
# Params for general matching
lowes_ratio: 0.8              # Ratio test for matches
matcher_type: FLANN           # FLANN, BRUTEFORCE, SIFT_GPU or WORDS
symmetric_matching: yes       # Match symmetricly or one-way
```
```matcher_type``` does not seem to be reliably over-ridden by the ODM option ```--matcher-type```

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
I would expect that any value passed to ```--matcher-type``` for ODM would successfully over-ride the ```matcher_type``` present in the OpenSFM config.py.

This is critical for proper operation of the ORB feature type, for instance (using BRUTEFORCE instead of FLANN).

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Pass ```--matcher-type bruteforce``` to a job with the default OpenSFM config.py value of FLANN in place, and watch it use FLANN for matching.
[image]

pierotofy, 2021-11-11:
Not a bug, it's just the way the logic is written: https://github.com/OpenDroneMap/ODM/blob/master/opendm/osfm.py#L217

closed 2021-11-11 by pierotofy

Saijin-Naib, 2021-11-11:
Would you accept a PR to force BRUTEFORCE for ORB/AKAZE for now?

pierotofy, 2021-11-12:
I'm a bit hesitant to include brute force as an option (due to its speed). There are no faster matchers?

pierotofy, 2021-11-15:
I'll add the necessary logic as part of #1362 :+1:

Saijin-Naib, 2021-11-15:
There are better matchers, but upstream OprnSFM doesn't expose them yet. Opencv has them, though.

I think as a first step this gets it mostly working.

We need Hamming Distance matchers for ORB/AKAZE (all binary descriptors in general), and I think I read we can (ab)use FLANN/kdtree with a special formula.

Thank you for hooking up the logic for this!