OpenDroneMap/ODM #1370: Implementation Issue: ORB incompatible with FLANN/KDTREE

Saijin-Naib, 2021-11-09:
### How did you install OpenDroneMap? (Docker, natively, ...)?
Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
NA

### What is the problem?
I implemented ORB improperly for how we have things set up currently. ORB and AKAZE (and all other binary feature descriptors) work best when not using a default FLANN/KDTREE matcher, but a binary-compatible matcher.

We might need conditional logic that disallows folks setting "broken" pairs of feature types to matcher types.

It also looks like #1368 comes into play here.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
When ORB and AKAZE are selected, the proper matcher type should be selected and used instead of the current way where a broken combination is used.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Use ODM with ORB and compare results to HAHOG/SIFT.

### Extra Information ###
https://forum.opencv.org/t/unsupported-error-when-using-orb-and-cv-flannbasedmatcher/5142/2?u=saijin_naib

https://docs.opencv.org/4.x/dc/dc3/tutorial_py_matcher.html

http://www.epacis.net/ccis2016/papers/paper_121.pdf

pierotofy, 2021-11-11:
We have three choices in OpenSfM:
 - FLANN
 - WORDS
 - BRUTEFORCE

BRUTEFORCE is not really practical though, because it's really slow.

Saijin-Naib, 2021-11-11:
I don't think so, no.

ORB/AKAZE with BRUTEFORCE are still far faster than SIFT+FLANN in most every case, and are far more robust than they were with FLANN.

[image]

I'm going to keep digging to see if BoW might be compatible with ORB/AKAZE, and to see if maybe other options might be tenable within OpenSFM since they are present within OpenCV which it is using.

pierotofy, 2021-11-12:
Ok, but here it's faster because there are `23` images :) It's a N^2 operation, so you will start to notice big slowdowns with ~60-80 images (and exponentially longer with more).

Saijin-Naib, 2021-11-14:
Took a few days to have my machine not bomb, but with a pagefile of 128GB (+32GB RAM) I got these results:  

---

* Defaults - ORB BRUTEFORCE - 524 images - 01:26:27
[image]

```
Created on: 11/13/2021, 11:08:36 PM
Processing Node: node-odm-1 (manual)
Options: auto-boundary: true, debug: true, dsm: true, dtm: true, feature-type: orb, matcher-type: bruteforce, verbose: true
Average GSD: 3.86 cm
Area: 1,797,537.35 m²
Reconstructed Points: 8,126,515
```
[image]
[image]
[image]
[Defaults-ORB-BRUTEFORCE-report.pdf](https://github.com/OpenDroneMap/ODM/files/7534527/Defaults-ORB-BRUTEFORCE-report.pdf)

----

* Defaults - SIFT FLANN - 524 images - 04:25:51
[image]

```
Created on: 11/14/2021, 12:41:28 AM
Processing Node: node-odm-1 (manual)
Options: auto-boundary: true, debug: true, dsm: true, dtm: true, verbose: true, rerun-from: dataset
Average GSD: 3.81 cm
Area: 1,995,760.47 m²
Reconstructed Points: 19,322,002
```
[image]
[image]
[image]
[Defaults-SIFT-FLANN-report.pdf](https://github.com/OpenDroneMap/ODM/files/7534528/Defaults-SIFT-FLANN-report.pdf)

I think given how SIFT is matching, forcing 15K ```--min-num-features``` for ORB would be a more equal comparison, but it is still illustrating the general trend, I think. I'm running that job now and will hopefully have results in about 2 hours.

pierotofy, 2021-11-15:
Is OpenSfM using brute force as a matcher (did you edit `osfm.py` to change the logic that writes the `config.yaml` file?) Seems strangely fast for brute force.

Saijin-Naib, 2021-11-15:
> Is OpenSfM using brute force as a matcher (did you edit `osfm.py` to change the logic that writes the `config.yaml` file?) Seems strangely fast for brute force.

Yep, absolutely did tweak that line in the OpenSFM config.py. Once I'm back home I'll attach the full processing logs for each job, as well.

You can tell if it does FLANN since it prints it in the console, and it also gets about 1/10th the number of points reconstructed as it does with BRUTEFORCE (which is also printed to the console).

With a proper matcher and tuning... I think we have some real potential here.

pierotofy, 2021-11-15:
Ohh, I see; I've actually misunderstood the algorithmic complexity of the brute force. It's still linear given that the number of ORB features is capped (is not influenced by the number of images, but by the number of features).

Well, this is good! Thanks for testing and providing runtime numbers.

#1369 needs a few tweaks, but I can help there if needed. The logic should make sure that the appropriate matcher is chosen in case ORB or AKAZE is used.

Saijin-Naib, 2021-11-15:
Awesome!

I too thought it was mostly image count.

Makes sense that it is feature pairs that causes the time difference, mostly.

Does the ORB feature being binary and 32bits VS vector and 64/128bits for SIFT also make sense that it would be able to have comparison and maths done on it faster?

I absolutely will need help. I'm sorry 😬

I will also need your assistance with implementing new matchers in #1368 once I research the most appropriate ones, and if they are in baseline OpenCV (I am certain they will be).

pierotofy, 2021-11-15:
So I ran some tests; the part I don't understand is why we can't continue using FLANN with both ORB/AKAZE. It seems to compute fine. I'm OK adding the brute force option, but I don't think we need to force the use of brute force for those two.

Saijin-Naib, 2021-11-15:
In my testing the binary feature descriptors were reconstructing about 1/2th the number of features with FLANN as they were with BRUTEFORCE. Speed was a bit faster, but yeah. That combined with bounded operation at 15K features max means they underperformed in reconstruction in a lot of datasets.

This image has a number of those tests:  
https://github.com/OpenDroneMap/ODM/issues/1370#issuecomment-966609270

Were you not finding that they were dropping half their matches with FLANN? Maybe OpenSFM updates or other changes mitigate that compared to what I have locally with 2.6.5?

pierotofy, 2021-11-15:
I guess I would expect the matcher to fail completely (crash or compute a wrong reconstruction) if it wasn't compatible.

I think I'll leave freedom to users to choose the matcher they want (with the exception of `bow`, which always requires hahog, or crash).

Saijin-Naib, 2021-11-15:
I'm just concerned with what I've read, like these:  
https://github.com/opencv/opencv/issues/5937#issuecomment-170083866  
https://answers.opencv.org/question/547/using-flann-with-binary-descriptors-brieforb/  

Even the OpenCV documentation seems to indicate the FLANN + Binary Descriptors needs some special magic:  
```
While using ORB, you can pass the following. The commented values are recommended as per the docs, but it didn't provide required results in some cases. Other values worked fine.:

FLANN_INDEX_LSH = 6
index_params= dict(algorithm = FLANN_INDEX_LSH,
                   table_number = 6, # 12
                   key_size = 12,     # 20
                   multi_probe_level = 1) #2
```
https://docs.opencv.org/4.x/dc/dc3/tutorial_py_matcher.html

pierotofy, 2021-11-15:
Mm, BRUTEFORCE in general should perform better than all others, because it compares all features with each other.

Either way, you can now choose `--matcher-type bruteforce`, which should do the trick.

Saijin-Naib, 2021-11-15:
Would the FLANN binary features tweak be something for a future Issue/PR then?

YanNoun, 2021-11-17:
Hi there,

Many thanks for the thorough experiments here !

Indeed, I would also expect ORB-Brute-Force to be lightning fast as Hamming distance for binary descriptor is usually SIMD-friendly and takes few cycles. 
Regarding FLANN+ORB, it is not expected to work as good as with real-valued descriptor (SIFT and friends), due to fundamental nature of binary-valued vector space.

Would be great if you would have any FLANN tweaking as the LSH use you mentioned.

Yann

pierotofy, 2021-12-18:
This should be fixed due to the latest improvements to OpenSfM (now included in ODM): https://github.com/mapillary/OpenSfM/commit/b5149ba981205eb18f82876f0e88595ab23e7031 :pray:

closed 2021-12-18 by pierotofy