OpenDroneMap/ODM #1372: ModuleNotFoundError in Dockerimage

sbonaime, 2021-11-17:
Debian 4.19.181-1 (2021-03-19) x86_64 GNU/Linux
 
### How did you install OpenDroneMap? (Docker, natively, ...)?

```docker pull opendronemap/odm

Using default tag: latest
latest: Pulling from opendronemap/odm
Digest: sha256:99de0d7bc0e438b79dbdfbf08bdf5bda4dc8fdb98c2d3fc7d8fb5ba7155624fd
Status: Image is up to date for opendronemap/odm:latest
docker.io/opendronemap/odm:latest
```

When I start odm with docker, I have the following error

```
docker run -it --rm -v  /data/images/my_image_dir/:/code/images -v "$(pwd)/odm_georeferencing:/code/odm_georeferencing"     -v "$(pwd)/odm_meshing:/code/odm_meshing"     -v "$(pwd)/odm_dem:/code/odm_dem"     -v "$(pwd)/odm_orthophoto:/code/odm_orthophoto"     -v "$(pwd)/odm_texturing:/code/odm_texturing"     -v "$(pwd)/odm_report:/code/odm_report" opendronemap/odm --orthophoto-png --mesh-octree-depth 12 --ignore-gsd --dtm   --smrf-threshold 0.4 --smrf-window 24 --dsm --pc-csv --pc-las --orthophoto-kmz   --ignore-gsd --feature-quality ultra --max-concurrency 16  --use-hybrid-bundle-adjustment --build-overviews --time --min-num-features 10000

Traceback (most recent call last):
  File "/code/run.py", line 10, in <module>
    from opendm import log
  File "/code/opendm/log.py", line 6, in <module>
    import dateutil.parser
ModuleNotFoundError: No module named 'dateutil'
```

I have the same problem with  opendronemap/nodeodm docker

sbonaime, 2021-11-17:
I just checked and I don't have the problem with opendronemap/odm:2.6.4 but it starts with 2.6.7

pierotofy, 2021-11-17:
Thanks @sbonaime :hand: this is indeed a bug; it looks like the build process failed to install the pip dependencies. (This is the second time this has happened, pypi might be experiencing some network issues as of late). I've made improvements to the build script, so updated and fixed images should be online shortly. :pray:

pierotofy, 2021-11-18:
Updated images are online, so this should be fixed (feel free to re-open if it's not). :+1:

closed 2021-11-18 by pierotofy

sbonaime, 2021-11-18:
@pierotofy So the bug was not in the code but rather in the docker process ?

originlake, 2021-11-18:
@pierotofy Is it possible to update the image with tag 2.6.7 as well? It's still not updated.

hyacz, 2022-07-25:
@pierotofy I have the same problem with [opendronemap/odm:gpu](https://hub.docker.com/layers/odm/opendronemap/odm/gpu/images/sha256-b2077e2b8333bb0825bbab245899ed4671e43d6bc58c3fd5340d07dae1406501?context=explore)
```
Job 41735 is waiting to be started...
Job 41735 has started on host h3cg01.
CUDA_VISIBLE_DEVICES=MIG-f11ed404-f6ed-5b7b-84af-2f21446f1959
===== RUN ODM =====
Traceback (most recent call last):
  File "/code/run.py", line 10, in <module>
    from opendm import log
  File "/code/opendm/log.py", line 6, in <module>
    import dateutil.parser
ModuleNotFoundError: No module named 'dateutil'
```