OpenDroneMap/ODM #1378: missing nvcuda.dll error

GHRKU, 2021-11-28:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Natively with ODM_Setup_2.6.7.exe

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 94.0.1
Windows 7 Pro 64-bit with SP1

### What is the problem?

Please see the issue reported for WebODM at
https://community.opendronemap.org/t/webodm-amd-gpu-program-exited-with-strange-error-code/9622

I have encountered the same issue running ODM (not WebODM) on the following hardware:

CPU: INTEL Core i7-7820X
RAM: 64 GB
GPU: Radeon RX 560

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

I would be grateful if you could implement a fix to to ODM to detect no CUDA drivers and switch to CPU as was done for WebODM (in v.1.9.11 Build 44), as well as any additional fixes that may be required per the above community post. 

Many thanks!

arashipl, 2021-11-30:
An additional information - the problem also happened to me on three separate machines:
i5 laptop + Intel HD4000 graphics
i7 desktop + NVidia HD4600 graphics
Core2Quad desktop + NVidia Quadro NVS510
The difference is that last one does not display nvcuda.dll error box but still fails on the same step. Same dataset is processed fine in ODM 2.6.4

pierotofy, 2021-11-30:
This should have been fixed with https://github.com/OpenDroneMap/ODM/pull/1374, we just didn't update the setup from the releases page. I'm re-running the build process, so the setup should get updated in the next few hours (try to re-download and re-install tomorrow).

GHRKU, 2021-12-02:
Thanks for updating the setup, @pierotofy. Is that update now complete? Is the new setup file located at the following URL?
https://github.com/OpenDroneMap/ODM/releases/download/v2.6.7/ODM_Setup_2.6.7.exe

Unfortunately I still get crashes after re-installing ODM with this setup file, however version 2.6.5 runs fine on the same machine. I am testing using drone_dataset_brighton_beach-master.

@arashipl, have you had a chance to try the new setup?

Thanks to you both.

pierotofy, 2021-12-03:
Mm, it should have completed (at least from the result of the GitHub actions); I didn't test it.

arashipl, 2021-12-03:
@GHRKU , no, I did not try again, github shows last update for assets for version 2.6.7 17 days ago, and it was mentioned within different thread that this will not likely be included before 2.6.8 is published - so I reverted to using 2.6.4.

pierotofy, 2021-12-03:
As a heads-up, there was a separate bug from the `missing nvcuda.dll` one, perhaps that's the cause of new crashes reported here.

I'm pushing another update right now.

GHRKU, 2021-12-04:
Thanks very much @pierotofy, the latest update works fine.

I really appreciate your help.

arashipl, 2021-12-06:
@pierotofy @GHRKU 
I have reinstalled currently availiable installer and I am happy to report that tileset processing finished without fail. Thank you!

closed 2021-12-06 by pierotofy