OpenDroneMap/ODM #1379: Possible Bug: Failure in mvstexturing (3221225477)

Saijin-Naib, 2021-11-29:
### How did you install OpenDroneMap? (Docker, natively, ...)?
Windows Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Windows 10 and Windows 11

### What is the problem?
Machines that have processed data succesfully with WebODM v1.9.11 Build 43 and below are now failing during the final texturing phase with Child Returned 3221225477. All machines seem to have lack of NVIDIA GPU and Windows native install as common factors. I can not reproduce with Windows Native Build 45 on Windows 11 with NVIDIA, nor can I reproduce with dockerized linux install on Intel iGPU. I do not have access to further hardware. Lightning is likewise unaffected.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
This codepath should not fail on non-NVIDIA systems. I can't figure out exactly where it is bombing, but during mvstexturing.

Saijin-Naib, 2021-11-30:
Current threads that seem to reproduce the behavior:  
https://community.opendronemap.org/t/processing-images-failed/9661?u=saijin_naib
https://community.opendronemap.org/t/new-user-question-strange-error-even-with-aukerman-image-set/9631/8?u=saijin_naib
https://community.opendronemap.org/t/processing-stopped-because-of-strange-values-in-the-reconstruction/9671?u=saijin_naib

pierotofy, 2021-11-30:
Thanks for compiling the list; it's definitely a recurring problem. Now if I can find a way to reproduce it... 😄

Saijin-Naib, 2021-12-01:
STR are enable Intel iGPU, disable/uninstall NVIDIA GPU and re-run dataset and it will fail:  
[image]

```
Created on: 11/30/2021, 6:16:17 PM
Processing Node: node-odm-1 (manual)
Options: auto-boundary: true, crop: 0, debug: true, dem-gapfill-steps: 4, dem-resolution: 1, dsm: true, dtm: true, feature-type: orb, matcher-neighbors: 16, mesh-size: 300000, min-num-features: 15000, orthophoto-resolution: 1, pc-classify: true, pc-filter: 1, pc-geometric: true, pc-quality: high, pc-sample: 0.01, resize-to: -1, use-3dmesh: true, verbose: true
```

[jvhooser-9623-intel-console.txt](https://github.com/OpenDroneMap/ODM/files/7629993/jvhooser-9623-intel-console.txt)

pierotofy, 2021-12-18:
Looking for a volunteer to lend their computer remotely so that I may try to reproduce it on the machine and debug it remotely. Get in touch if you're witnessing this bug!

closed 2023-02-18 by Saijin-Naib