OpenDroneMap/ODM #1380: Feature Request:  Prevent system sleep during ODM processing

Saijin-Naib, 2021-12-01:
### How did you install OpenDroneMap? (Docker, natively, ...)?
NA

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Windows
MacOS
Linux

### What is the problem?
Currently, the OS' sleep settings can cause it to fall to sleep during long-running ODM processing jobs.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
Ideally, while actively processing, ODM would use platform-native methods to keep the system awake until the ODM processing completes or fails, then releasing the wakelock.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Proceess a task that takes longer than the system sleep settings without any user intervention. The machine will fall to sleep.

### Additional Context ###
https://github.com/np-8/wakepy

pierotofy, 2021-12-01:
I'm inclined to say that ODM cannot/should not be in charge of preventing system sleep (it would be really messy to implement, if possible at all, and would not work if you use docker).

Saijin-Naib, 2021-12-01:
Ah... Okay.

Windows has a public API for programs to say "hey, don't sleep right now I'm busy" that many things take advantage of.

The user can override this by manually choosing to sleep or pressing the power button, so it isn't something dangerous, but a convenience.

MacOS has a similar built-in tool with caffeinate which can be called from the terminal.

I use similar under Linux via GNOME.

closed 2021-12-07 by pierotofy