OpenDroneMap/ODM #1381: Possible Bug: Too many indicies for array in OpenSFM features.py

Saijin-Naib, 2021-12-01:
### How did you install OpenDroneMap? (Docker, natively, ...)?
Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Alpine Linux Edge with linux-lts (5.15.6-r0)

### What is the problem?
When processing 320x255 FLIR data locally and on lightning, ORB is crashing with the following error:  
```
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed
```

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
ORB feature extraction should not fail.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
Log attached. Data shared to reproduce the issue are private.
[asa_f9406-ORB-console.txt](https://github.com/OpenDroneMap/ODM/files/7636989/asa_f9406-ORB-console.txt)

pierotofy, 2022-01-04:
Is this still an issue since the latest OpenSfM update? Would be good to retry to see if this was fixed.

Saijin-Naib, 2022-01-04:
No idea. Let me re-try with that data and get back to you.

Would the Windows v1.9.11 Build 49 be sufficient to test with?

pierotofy, 2022-01-04:
> Would the Windows v1.9.11 Build 49 be sufficient to test with?

Yes :+1:

Saijin-Naib, 2022-01-12:
Waiting on getting the dataset again. I will follow up soon.

sbonaime, 2022-03-04:
Hi,
I have the same error with my last thermal dataset and latest ODM Docker image. With SIFT, I don't have the problem.  Here is the ORB log
````
2022-03-04 10:05:59,767 DEBUG: Computing ORB
2022-03-04 10:05:59,797 DEBUG: Found 2253 points in 0.0289304256439209s
2022-03-04 10:05:59,803 DEBUG: No segmentation for DJI_0589_R.JPG, no features masked.
2022-03-04 10:05:59,826 DEBUG: Found 9189 points in 0.0809636116027832s
2022-03-04 10:05:59,830 DEBUG: No segmentation for DJI_0407_R.JPG, no features masked.
Traceback (most recent call last):
  File "/code/SuperBuild/install/bin/opensfm/bin/opensfm_main.py", line 25, in <module>
    commands.command_runner(
  File "/code/SuperBuild/install/bin/opensfm/opensfm/commands/command_runner.py", line 37, in command_runner
    command.run(data, args)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/commands/command.py", line 12, in run
    self.run_impl(data, args)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/commands/detect_features.py", line 11, in run_impl
    detect_features.run_dataset(dataset)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/actions/detect_features.py", line 15, in run_dataset
    features_processing.run_features_processing(data, data.images(), False)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/features_processing.py", line 65, in run_features_processing
    parallel_map(process, arguments, processes, 1)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/context.py", line 56, in parallel_map
    res = Parallel(batch_size=batch_size)(delayed(func)(arg) for arg in args)
  File "/usr/local/lib/python3.9/dist-packages/joblib/parallel.py", line 1061, in __call__
    self.retrieve()
  File "/usr/local/lib/python3.9/dist-packages/joblib/parallel.py", line 940, in retrieve
    self._output.extend(job.get(timeout=self.timeout))
  File "/usr/lib/python3.9/multiprocessing/pool.py", line 771, in get
    raise self._value
  File "/usr/lib/python3.9/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/usr/local/lib/python3.9/dist-packages/joblib/_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/joblib/parallel.py", line 262, in __call__
    return [func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/joblib/parallel.py", line 262, in <listcomp>
    return [func(*args, **kwargs)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/features_processing.py", line 120, in process
    run_detection(queue)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/features_processing.py", line 155, in run_detection
    detect(image, image_array, segmentation_array, instances_array, data, force)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/features_processing.py", line 227, in detect
    p_unmasked, f_unmasked, c_unmasked = features.extract_features(
  File "/code/SuperBuild/install/bin/opensfm/opensfm/features.py", line 568, in extract_features
    xs = points[:, 0].round().astype(int)
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
  File "/code/stages/odm_app.py", line 94, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 346, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 346, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 346, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 327, in run
    self.process(self.args, outputs)
  File "/code/stages/run_opensfm.py", line 35, in process
    octx.feature_matching(self.rerun())
  File "/code/opendm/osfm.py", line 322, in feature_matching
    self.run('detect_features')
  File "/code/opendm/osfm.py", line 34, in run
    system.run('"%s" %s "%s"' %
  File "/code/opendm/system.py", line 106, in run
    raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[91m[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/[0m
````

sbonaime, 2022-03-04:
https://github.com/mapillary/OpenSfM/issues/849

pierotofy, 2022-07-07:
This should be fixed since the original issue in OpenSfM has been fixed (and those changes are now in ODM). :clinking_glasses: 

Feel free to re-open if the issue persists.

closed 2022-07-07 by pierotofy

Saijin-Naib, 2022-07-07:
Indeed, processes right through successfully now!

Thanks so much!